smartdevicelink/sdl_core #1895: SDL behavior in case 'HeartBeat' param is empty

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL behavior in case 'HeartBeat' param is empty

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. "HeartBeat" param is empty in .ini file 

## Actual result
N/A

## Expected result
SDL must set the default value = 5000 ms
respond ACK to HeartBeat_request from mobile app
send HeartBeat_request to mobile app 

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Low since there is no significant functionality loss and default value of 'HeartBeatTimeout' parameter is present in .INI file

robinmk, 2018-06-21:
1. What is the actual result?
2. Is the Mobile App initiating this flow by triggering a heart_beat request?

dboltovskyi, 2018-06-22:
@robinmk 
1. Actual result is provided in updated description 
2. Correct - Mobile App initiates heartbeat flow

iCollin, 2021-04-15:
Closed via https://github.com/smartdevicelink/sdl_core/pull/2513

PR #2513 (2021-04-15): Change default HeartBeat value
Fixes #1895 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
kDefaultHeartBeatTimeout = 5000ms

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2021-04-15 by iCollin