smartdevicelink/sdl_core #1896: SDL behavior in case 'HeartBeat' param is zero

LitvinenkoIra, 2017-11-23:
**Occurrence: Always** 

# Description
SDL behavior in case 'HeartBeat' param is zero

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. "HeartBeat" param has value "0" (zero) in .ini file

## Actual result
N/A

## Expected result
SDL must respond ACK to HeartBeat_request from mobile app
SDL must NOT send HeartBeat_request to mobile app 

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Low since there is no significant functionality loss and default value of 'HeartBeatTimeout' parameter is present in .INI file

ychernysheva, 2020-05-22:
This issue is not reproducible on release/6.1.0 branch