smartdevicelink/sdl_core #1897: SDL does not resume mobile app after unexpected disconnect due to DEACTIVATE_HMI reason

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
SDL does not resume mobile app after unexpected disconnect due to DEACTIVATE_HMI reason

## Preconditions
 - SDL and HMI are started.
 - App is registered and activated.

## Steps to reproduce
1. Mobile app running on SDL in FULL or LIMITED
and re-registers after unexpected disconnect due to DEACTIVATE_HMI mode activation during the same ignition cycle
and HMILevel resumption was postponed for this app (APPLINK-16153)
and mobile app re-registers after unnexpected disconnect due to DEACTIVATE_HMI mode deactivation during the same ignition cycle.

## Actual result
SDL does not resume mobile app.

## Expected result
SDL must resume HMILevel for this app being before DEACTIVATE_HMI activation (by sending OnHMIStatus notification to this media app).

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium since issue is related to a specific case and only for HMILevel notification

HSavynetska, 2018-09-06:
Defect is not actual.
Mobile app is not unexpected disconnect due to DEACTIVATE_HMI.

closed 2019-09-27 by mked-luxoft