smartdevicelink/sdl_core #1898: Сanceling HMILevel resumption of media app after unexpectedDisconnect and re-registration during active embedded audio source

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
SDL does not resume mobile app after unexpected disconnect due to DEACTIVATE_HMI reason

## Preconditions
 -  SDL and HMI are started.
 -  Media app is registered.

## Steps to reproduce
1. Media app being in FULL or LIMITED and AUDIBLE unexpectedly disconnects and re-registers during active embedded audio source within the same ignition cycle and SDL does NOT receive OnEventChanged (AUDIO_SOURCE, isActive=false) from HMI during ApplicationResumingTimeout (the value defined at .ini file)

## Actual result
SDL does not cancel HMILevel resumption.

## Expected result
SDL must cancel HMILevel resumption for this media app (meaning: media app must be in NONE)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium since issue is related to a specific case and only for HMILevel notification

mkorniichuk, 2018-09-19:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2069
SDL build: develop branch ( 28235f88a6733106a7e8a0d490dfb16a5f374a33 )
ATF: develop branch

Status: Defect is not reproduced on current develop.

IGapchuk, 2019-09-24:
Environment:
Script: smartdevicelink/sdl_atf_test_scripts#2069
SDL build: develop (6b43b00cdcafb5bc386a9fac603b596b882b4c39)
ATF: develop (https://github.com/smartdevicelink/sdl_atf/commit/b416694fb67de93ae80888a0d689b636e59f53f7)

Status: Defect is not reproduced on current develop.

closed 2019-09-24 by IGapchuk