smartdevicelink/sdl_core #1899: SDl does not resume media app after unexpectedDisconnect and re-registration during active embedded audio source

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
SDl does not resume media app after unexpectedDisconnect and re-registration during active embedded audio source.

## Preconditions
 -  SDL and HMI are started.
 -  Media app is registered.

## Steps to reproduce
1. Media app being in FULL or LIMITED and AUDIBLE unexpectedly disconnects
and re-registers during active embedded audio source within the same ignition cycle
and SDL receives OnEventChanged (AUDIO_SOURCE, isActive=false) from HMI during ApplicationResumingTimeout (the value defined at .ini file)

## Actual result
SDL does not resume HMILevel and audioStreamingState.

## Expected result
SDL must resume HMILevel and audioStreamingState of this media app being before unexpected disconnect (by sending OnHMIStatus notification to this media app).

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium since issue is related to a specific case and only for HMILevel notification

HSavynetska, 2019-11-22:
The defect is not reproduced on the current develop

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2077
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

closed 2019-11-22 by HSavynetska