smartdevicelink/sdl_core #1900: Media app does not get AUDIBLE if it's successfully resuming during active embedded navigation

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Media app does not get AUDIBLE if it's successfully resuming during active embedded navigation

## Preconditions
 -  SDL and HMI are started.

## Steps to reproduce
1. Media app registers during active embedded navigation and satisfies all conditions for HMILevel resumption.

## Actual result
SDL does not resume HMILevel and audioStreamingState.

## Expected result
SDL must set AUDIBLE audioStreamingState to this media app and send via OnHMIStatus together with resumed HMILevel (FULL or LIMITED).

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium since issue is related to a specific case and only for HMILevel notification

Ypostolov, 2018-09-17:
Environment:

SDL build: develop branch ( 28235f88a6733106a7e8a0d490dfb16a5f374a33 )
HMI: develop branch ( 5fc453a64b9962a66e203299e5a0ac57bb62541d )
SPT: 20180801-Nightly-Android

Status: Defect is not reproduced.

mked-luxoft, 2019-09-27:
SDL : e356886
ATF : 05ba940

not reproducible

closed 2019-09-27 by mked-luxoft