smartdevicelink/sdl_core #1902: Media app and navi app both must get AUDIBLE streaming state after successfull resumption

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Media app and navi app both must get AUDIBLE streaming state after successfull resumption

## Preconditions
 -  SDL and HMI are started.

## Steps to reproduce
- Media app registers and gets FULL/LIMITED and AUDIBLE
- Navigation app registers and satisfies all conditions for HMILevel resumption

## Actual result
SDL does not resume HMILevel and audioStreamingState of navigation app.

## Expected result
SDL must resume navi app to appropriate <HMILevel> (according to existing req-s) and AUDIBLE audioStreamingState (by sending OnHMIStatus notification to mobile app).

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium since issue is related to a specific case and only for HMILevel notification

Ypostolov, 2018-09-28:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2072
SDL build: develop branch https://github.com/smartdevicelink/sdl_core/commit (28235f88a6733106a7e8a0d490dfb16a5f374a33 )
ATF: develop branch

Status: Defect is not reproducing in current develop

HSavynetska, 2019-11-22:
The defect is not reproduced on the current develop

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2072
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

closed 2019-11-22 by HSavynetska