smartdevicelink/sdl_core #1903: Media app is not activated during active embedded navigation

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Media app is not activated during active embedded navigation.

## Preconditions
 -  SDL and HMI are started.

## Steps to reproduce
- Media app in LIMITED and AUDIBLE due to active embedded navigation
- SDL receives SDL.ActivateApp (<appID_of_media_app>) from HMI

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
SDL must respond SDL.ActivateApp (SUCCESS) to HMI send OnHMIStatus (FULL, AUDIBLE) to mobile app (embedded navigation is still active).

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for media application during active embedded navigation.

HSavynetska, 2019-11-22:
The defect is reproduced on the current develop

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2074
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@GetmanetsIrina can you please advise if you are able to reproduce this issue in the latest Core release?

LuxoftAKutsan, 2020-05-26:
Issue reproduced on 0ce76256232bad18bac986e7e68f35c2ae510a29 HEAD, tag: 6.1.1

theresalech, 2020-05-26:
@LuxoftAKutsan thank you for the additional information.  Can you please also advise if Luxoft will be submitting a PR to resolve this issue?

dboltovskyi, 2020-06-04:
Updated ATF script and logs: [2020-06-04_15-47-08_1903.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4732392/2020-06-04_15-47-08_1903.tar.gz)

theresalech, 2020-06-08:
@dboltovskyi thank you for the update!  Can you please advise if Luxoft is planning to submit a PR to resolve this issue?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

jacobkeeler, 2020-09-04:
Closing with the merge of #3476

closed 2020-09-04 by jacobkeeler