smartdevicelink/sdl_core #1904: Media app activation during active embedded audio source

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Media app activation during active embedded audio source

## Preconditions
 -  SDL and HMI are started.
 - Media app is registered.
 - Media app in BACKGROUND and NOT_AUDIBLE due to active embedded audio source.

## Steps to reproduce
- User activates this media app and SDL receives from HMI:
a) OnEventChanged (AUDIO_SOURCE, isActive=false)
b) SDL.ActivateApp (<appID_of_communication_app>) 

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
SDL must respond SDL.ActivateApp (SUCCESS) to HMI send OnHMIStatus (FULL, AUDIBLE) to mobile app.
Information:
1. Expected HMI behavior:
a. Media app activation is the trigger for HMI to switch off embedded audio source
b. HMI switches off embedded audio source and sends OnEventChanged (AUDIO_SOURCE, isActive=false) to SDL

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for media application during active embedded audio source.

HSavynetska, 2019-11-27:
The issue is not relevant on the current develop branch

Environment:
 Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2070
 SDL build: [https://github.com/smartdevicelink/sdl_core/tree/develop] ( e137556 )
 ATF: develop branch

closed 2019-11-27 by HSavynetska