smartdevicelink/sdl_core #1906: Navigation app must be activated during active embedded navigation and "MixingAudioSupported" = true

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Navigation app must be activated during active embedded navigation and "MixingAudioSupported" = true.

## Preconditions
- MixingAudioSupported" = true at .ini file
- SDL and HMI are started.
- Navigation app is registered.
- Navigation app in LIMITED and AUDIBLE due to active embedded audio source

## Steps to reproduce
- SDL receives SDL.ActivateApp (<appID_of_navigation_app>) from HMI

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
 SDL must send OnHMIStatus (FULL, AUDIBLE) to mobile app.

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for Navigation application during active embedded navigation.

HSavynetska, 2018-07-17:
**Verified**:
**SDL build**: develop branch ( 45109a0 )
**ATF**: develop branch

**Status**: Defect is reproduced.

To current requirement item 5.1 in https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0150-video-streaming-state.md( audioStreamingState must be "NOT_AUDIBLE" due/during to active embedded audio source) 

But in the precondition described steps lead to new sdl defect:

Actual result
Navigation app in BACKGROUND and NOT_AUDIBLE due to active embedded audio source

Expected result
Navigation app in LIMITED and NOT_AUDIBLE due to active embedded audio source

Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1977

HSavynetska, 2019-11-21:
The defect is reproduced on the current develop

Environment:
 Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1977
 SDL build: [https://github.com/smartdevicelink/sdl_core/tree/develop] (891497b )
 ATF: develop branch

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Can the reproduction steps described (or this test script) actually reproduce this issue? Also, Does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@HSavynetska, can you please advise if you are seeing this issue when using Core 6.1?

dboltovskyi, 2020-06-03:
@theresalech Please notice issue is still actual for Core 6.1
ATF script and logs: [2020-06-02_20-50-16_1906.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4720430/2020-06-02_20-50-16_1906.tar.gz)

theresalech, 2020-06-03:
Hi @dboltovskyi, thank you for the additional information.  Will Luxoft be submitting a PR to resolve this issue?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

jacobkeeler, 2020-07-27:
Duplicate of #1608

closed 2020-07-27 by jacobkeeler