smartdevicelink/sdl_core #1907: Non-media app must be activated during active embedded navigation

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Non-media app must be activated during active embedded navigation

## Preconditions
- SDL and HMI are started.
- Non-media app is registered.
- Non-media app in BACKGROUND and NOT_AUDIBLE due to active embedded navigation

## Steps to reproduce
- SDL receives SDL.ActivateApp(<appID_of_non-media_app>) from HMI

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
 SDL must respond SDL.ActivateApp (SUCCESS) to HMI send OnHMIStatus (FULL, NOT_AUDIBLE).

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for non-media application during active embedded navigation.

HSavynetska, 2018-07-12:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1967
SDL build: develop branch ( b7da3b5 )
ATF: develop branch

Status: defect is reproduced.

HSavynetska, 2019-11-21:
The defect is reproduced on the current develop

Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1967
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop (891497b )
ATF: develop branch

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@GetmanetsIrina can you please advise if you are able to reproduce this issue in the latest Core release?

dboltovskyi, 2020-06-04:
@theresalech Please notice issue is still actual for Core 6.1
ATF script and logs: [2020-06-04_15-05-19_1907.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4732222/2020-06-04_15-05-19_1907.tar.gz)

theresalech, 2020-06-08:
@dboltovskyi thank you for the update!  Can you please advise if Luxoft is planning to submit a PR to resolve this issue?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

jacobkeeler, 2020-09-04:
Closing with the merge of #3476

closed 2020-09-04 by jacobkeeler