smartdevicelink/sdl_core #1908: Non-media app must be activated during active audio source

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Non-media app must be activated during active audio source

## Preconditions
- SDL and HMI are started.
- Non-media app is registered.
- Non-media app in BACKGROUND and NOT_AUDIBLE due to active embedded audio source

## Steps to reproduce
- SDL receives SDL.ActivateApp(<appID_of_non-media_app>) from HMI

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
 SDL must respond SDL.ActivateApp (SUCCESS) to HMI send OnHMIStatus (FULL, NOT_AUDIBLE).

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for non-media application during active audio source.

HSavynetska, 2018-07-12:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1966
SDL build: develop branch ( b7da3b5 )
ATF: develop branch

Status: defect is not reproduced.

HSavynetska, 2019-11-28:
The issue is relevant on the current develop branch

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1966
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( e137556 )
ATF: develop branch

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this issue. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this issue.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is important for Toyota. Therefore, we would appreciate it if you can make this a high-priority.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

jacobkeeler, 2020-09-04:
Closing with the merge of #3476

closed 2020-09-04 by jacobkeeler