smartdevicelink/sdl_core #1909: SDL must set LIMITED HMILevel in case app was activated during postponed HMILevel resumption

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
SDL must set LIMITED HMILevel in case app was activated during postponed HMILevel resumption

## Preconditions
- SDL and HMI are started.
- Media app is registered.

## Steps to reproduce
- Media app is resuming after unexpected disconnect.
- HMILevel resumption of media app was postponed and SDL receives SDL.ActivateApp (<appID_of_media_app>).

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
 SDL must respond SDL.ActivateApp(SUCCESS) to HMI resume HMILevel of media app after event ends activate app and send OnHMIStatus (LIMITED, AUDIBLE) to this media app.
Information: non-media apps:
a. can be activated during active phone call
b. can not be activated during OnEmergencyEvent (enabled=true) or active VR session (as there is no way to activate it during these events)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium since issue is related to a specific case and only for HMILevel notification

HSavynetska, 2019-11-27:
The issue is not relevant on the current develop branch
Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2068
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

closed 2019-11-27 by HSavynetska