smartdevicelink/sdl_core #1910: Navigation app must get LIMITED and AUDIBLE in case embedded audio source is activated and "MixingAudioSupported" = true

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Navigation app must get LIMITED and AUDIBLE in case embedded audio source is activated and "MixingAudioSupported" = true

## Preconditions
- "MixingAudioSupported" = true at .ini file.
- SDL and HMI are started.
- Navigation app is registered.

## Steps to reproduce
-  Navigation app in FULL or LIMITED and AUDIBLE and SDL receives OnEventChanged (AUDIO_SOURCE, isActive=true) from HMI.

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
SDL must send OnHMIStatus (LIMITED, AUDIBLE) to mobile app.

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for Navigation application during active audio source.

dboltovskyi, 2020-06-15:
Even if the issue contradicts to the item `5.1` in https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0150-video-streaming-state.md we suppose it's still actual for `6.1` release.
Please find updated ATF script and logs: [2020-06-15_12-32-24_1910.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4781652/2020-06-15_12-32-24_1910.tar.gz)

jacobkeeler, 2020-09-09:
Fixed by #3476

closed 2020-09-09 by jacobkeeler