smartdevicelink/sdl_core #1911: SDL must set media app to LIMITED and AUDIBLE due to active embedded navigation

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
SDL must set media app to LIMITED and AUDIBLE due to active embedded navigation

## Preconditions
- SDL and HMI are started.
- Media app is registered and activated.

## Steps to reproduce
-  Media app in FULL and AUDIBLE and SDL receives OnEventChanged (EMBEDDED_NAVI, isActive=true) from HMI.

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
SDL must send OnHMIStatus (LIMITED, AUDIBLE) to mobile app.

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for media application during active embedded navigation.

HSavynetska, 2018-07-03:
SDL build: develop branch (ba1da7b)
ATF: develop branch

Status: defect is not reproduced.

AStasiuk, 2018-07-20:
Closed as not actual.

closed 2018-07-20 by AStasiuk