smartdevicelink/sdl_core #1912: SDL must respond NACK in case navigation app connected over protocol v2 sends StartService for audio service

LitvinenkoIra, 2017-11-24:
**Occurrence: Always** 

# Description
SDL must respond NACK in case navigation app connected over protocol v2 sends StartService for audio service

## Preconditions
 - SDL and HMI are started.

## Steps to reproduce
1. navigation app connects over protocol v2 and this app sends StartService for audio service

## Actual result
N/A

## Expected result
SDL must respond StartService_NACK to this app

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to High. Issue may cause a problems with streaming for applications which is using protocol v2.

GetmanetsIrina, 2017-12-28:
Issue is reproduced on develop branch(0b19cf498d1c9972d67b85793dc6b029c26378a0)

jacobkeeler, 2018-01-31:
Verified that this is fixed by #1971.

jacobkeeler, 2018-01-31:
Fixed with the merge of #1971

closed 2018-01-31 by jacobkeeler