smartdevicelink/sdl_core #1913: SDL must set NOT_AUDIBLE to non-media app in case this app is activated during phone call

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
SDL must set NOT_AUDIBLE to non-media app in case this app is activated during phone call

## Preconditions
- SDL and HMI are started.
- Non-media app is registered.

## Steps to reproduce
-  SDL receives SDL.ActivateApp(<appID_of_non-media_app>) request from HMI during phone call (non-media -> app that does NOT use audio channel).

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
SDL must respond SDL.ActivateApp (SUCCESS) to HMI send OnHMIStatus ("HMILevel: FULL, audioStreamingState: NOT_AUDIBLE") to requested for activation app.

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority  was set to Medium. Issue is related to setting wrong  audioStreamingState for activated application in some specific cases.  May affect correct permissions calculation for non-media application  during phone call.

HSavynetska, 2018-07-09:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1964
SDL build: develop branch (f80e75e)
ATF: develop branch

Status: defect is not reproduced.

AStasiuk, 2018-07-20:
Closed as not reproduced.

closed 2018-07-20 by AStasiuk