smartdevicelink/sdl_core #1914: SDL must set NOT_AUDIBLE to media app in case this app is activated during phone call

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
SDL must set NOT_AUDIBLE to media app in case this app is activated during phone call

## Preconditions
- SDL and HMI are started.
- Media app is registered.

## Steps to reproduce
-  SDL receives SDL.ActivateApp(<appID_of_media_app>) request from HMI during phone call (or <appID_of_navigation_app>, <appID_of_communication_app>)

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
SDL must respond SDL.ActivateApp (SUCCESS) to HMI
send OnHMIStatus ("HMILevel: FULL, audioStreamingState: NOT_AUDIBLE") to requested for activation app
send OnHMIStatus ("HMILevel: FULL, audioStreamingState: AUDIBLE") to requested-to-activate-app after the phone call ends SDL receives OnEventChanges (PHONE_CALL, isActive=false) from HMI)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for media application during phone call.

HSavynetska, 2018-07-03:
SDL build: develop branch (ba1da7b)
ATF: develop branch

Status: defect is not reproduced.

AStasiuk, 2018-07-20:
Closed as not actual

closed 2018-07-20 by AStasiuk