smartdevicelink/sdl_core #1915: Media app must get ATTENUATED streaming state when navi app starts streaming

LitvinenkoIra, 2017-11-24:
**Occurrence: Always** 

# Description
Media app must get ATTENUATED streaming state when navi app starts streaming

## Preconditions
 - SDL and HMI are started.

## Steps to reproduce
1. media app is running in FULL (or LIMITED) and AUDIBLE at SDL
and navigation app is running in LIMITED (or FULL when no any apps in FULL now) and AUDIBLE on SDL
and navigation app starts streaming
and "MixingAudioSupported" = true at .ini file

## Actual result
N/A

## Expected result
SDL must send OnHMIStatus (<current_HMILevel>, ATTENUATED) to media app (and return media app to AUDIBLE right after navi app stops streaming)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for media application during navigation application streaming.

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this issue. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this issue.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

GetmanetsIrina, 2020-04-29:
The issue is covered by https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/MobileProjection/Phase2/025_Media_app_must_get_ATTENUATED_streaming_state_when_navi_app_starts_streaming.lua

iCollin, 2020-05-18:
Hello, I cannot reproduce this issue against [`sdl_core@6.1.1`](https://github.com/smartdevicelink/sdl_core/tree/6.1.1).

[025_Media_app_must_get_ATTENUATED_streaming_state_when_navi_app_starts_streaming.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/MobileProjection/Phase2/025_Media_app_must_get_ATTENUATED_streaming_state_when_navi_app_starts_streaming.lua) test passes against master once [the incorrect expectation](https://github.com/smartdevicelink/sdl_atf_test_scripts/issues/2404) is rectified. Thank you @GetmanetsIrina for identifying this script and its issue.

I am closing this issue as it is already fixed.

closed 2020-05-18 by iCollin