smartdevicelink/sdl_core #1916: Media app: system supports audio mixing and embedded navigation starts streaming (TTS session)

LitvinenkoIra, 2017-11-24:
**Occurrence: Always** 

# Description
Media app: system supports audio mixing and embedded navigation starts streaming (TTS session)

## Preconditions
 - SDL and HMI are started.
- "MixingAudioSupported" = true at .ini file

## Steps to reproduce
1. media app in LIMITED and AUDIBLE due to active embedded navigation 
2. SDL receives TTS.Started from HMI (embedded navigation starts streaming)

## Actual result
N/A

## Expected result
SDL must send OnHMIStatus (LIMITED, ATTENUATED) to mobile app
send OnHMIStatus (LIMITED, AUDIBLE) to mobile app right after embedded navigation stops streaming (SDL receives TTS.Stopped from HMI)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for media application during audio streaming.

HSavynetska, 2019-11-28:
The issue is relevant on the current develop branch
Environment:
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( e137556 )
ATF: develop branch
Test script:
[1916_System_supports_audio_mixing_and_embedded_navigation_starts_streaming_TTS session.zip](https://github.com/smartdevicelink/sdl_core/files/3901666/1916_System_supports_audio_mixing_and_embedded_navigation_starts_streaming_TTS.session.zip)
Test script was created on the current develop branch (commit: 6f43bfd)

GetmanetsIrina, 2020-04-27:
Script is updated according to changes in ATF:
ATF: https://github.com/smartdevicelink/sdl_atf/commit/6f0aff7bef8ba7f8c3adbcbf04683862c03128f5
Script: [1916_System_supports_audio_mixing_and_embedded_navigation_starts_streaming_TTS_session.zip](https://github.com/smartdevicelink/sdl_core/files/4540324/1916_System_supports_audio_mixing_and_embedded_navigation_starts_streaming_TTS_session.zip)

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@LitvinenkoIra can you please advise if you are able to reproduce this issue in the latest Core release?

dboltovskyi, 2020-06-04:
@theresalech Please notice issue is still actual for Core 6.1
ATF script and logs: [2020-06-04_15-38-16_1916.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4732359/2020-06-04_15-38-16_1916.tar.gz)

theresalech, 2020-06-08:
@dboltovskyi thank you for the update!  Can you please advise if Luxoft is planning to submit a PR to resolve this issue?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

jacobkeeler, 2020-09-04:
Closing with the merge of #3476

PR #3476 (2020-09-04): Overhaul state manager to handle regular state changes during hmi events
Fixes #1916, #1917, #1918, #1920, #1908, #1608, #1907, #1903, #2322, #2227, #3167 

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
Add unit tests and ATF tests covering this overhaul

### Summary
The implementation of the state manager did not allow for state changes above a certain level during several different HMI events (such as AUDIO_SOURCE and EMBEDDED_NAVI). These changes add a way of separating the maximum HMI state from the initial HMI state for an event, and all [... cut]
closed 2020-09-04 by jacobkeeler