smartdevicelink/sdl_core #1918: Multiple media and navigation apps activation during active embedded audio source+audio mixing supported

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Multiple media and navigation apps activation during active embedded audio source+audio mixing supported

## Preconditions
- "MixingAudioSupported" = true at .ini file
- SDL and HMI are started.
- Three apps running on system:
 a. media app_1 in FULL and AUDIBLE
 b. navigation app_2 in LIMITED and AUDIBLE
 c. non-media app_3 in BACKGROUND and NOT_AUDIBLE
- User activates embedded audio source and HMILevel of apps were changed to: 
 a. media app_1 in BACKGROUND and NOT_AUDIBLE
 b. navigation app_2 in LIMITED and AUDIBLE

## Steps to reproduce
- User activates navigation app_2
HMI -> SDL: SDL.ActivateApp ( <appID_2> )
SDL -> HMI: SDL.ActivateApp (SUCCESS)
- SDL -> navigation app_2: OnHMIStatus (FULL, AUDIBLE)
Note: embedded audio source is still active
- User activates media app_1
HMI -> SDL: OnEventChanged (AUDIO_SOURCE, isActive:false)
HMI -> SDL: OnAppDeactivated ( <appID_2> )
HMI -> SDL: SDL.ActivateApp ( <appID_1 )

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
1. SDL -> navigation app_2 : OnHMIStatus (LIMITED, AUDIBLE)
2. SDL -> HMI: SDL.ActivateApp (SUCCESS)
3. SDL -> media app_1 : OnHMIStatus (FULL, AUDIBLE)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for navi activated application.

HSavynetska, 2019-11-28:
The issue is relevant on the current develop branch

Environment:
Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2089
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( e137556 )
ATF: develop branch

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Can the reproduction steps described actually reproduce this issue? Also, Does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@HSavynetska, can you please advise if you are seeing this issue when using Core 6.1?

dboltovskyi, 2020-06-03:
@theresalech Please notice issue is still actual for Core 6.1
ATF script and logs: [2020-06-02_20-52-08_1918.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4720433/2020-06-02_20-52-08_1918.tar.gz)

theresalech, 2020-06-03:
Hi @dboltovskyi, thank you for the additional information. Will Luxoft be submitting a PR to resolve this issue?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

jacobkeeler, 2020-09-04:
Closing with the merge of #3476

closed 2020-09-04 by jacobkeeler