smartdevicelink/sdl_core #1919: Multiple media and navigation apps activation during active embedded audio source+audio mixing supported

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Multiple media and navigation apps activation during active embedded audio source+audio mixing supported

## Preconditions
- "MixingAudioSupported" = true at .ini file
- SDL and HMI are started.
- Three apps running on system:
  a. media app_1 in FULL and AUDIBLE
  b. navigation app_2 in LIMITED and AUDIBLE
  c. non-media app_3 in BACKGROUND and NOT_AUDIBLE
- User activates embedded navigation and HMILevel of apps were changed to:
  a. media app_1 in LIMITED and AUDIBLE
  b. navigation app_2 in BACKGROUND and NOT_AUDIBLE
- User activates media app_1 -> FULL, AUDIBLE
  a. navigation app_2 still in BACKGROUND and NOT_AUDIBLE
  b. non-media app_3 still in BACKGROUND and NOT_AUDIBLE
- User activates navigation app_2 -> FULL, AUDIBLE

Note: embedded navigation was switched off
media app_1 gets (LIMITED, AUDIBLE)
-> non-media app_3 still in BACKGROUND and NOT_AUDIBLE

## Steps to reproduce
-  User activates non-media app_3
HMI -> SDL: OnAppDeactivated ( <appID_2> )
HMI -> SDL: SDL.ActivateApp ( <appID_3 )

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
1. SDL -> navigation app_2 : OnHMIStatus (LIMITED, AUDIBLE)
2. SDL -> HMI: SDL.ActivateApp (SUCCESS)
3. SDL -> non-media app_3 : OnHMIStatus (FULL, NOT_AUDIBLE)
Note: media app_1 still in LIMITED and AUDIBLE

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for non-media activated application.

HSavynetska, 2019-12-13:
The issue is not relevant on the current develop branch.
The specified requirements in the issue are not relevant.
You can find the current requirements for the changes in HMILevel and audioStreamingState here
https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0150-video-streaming-state.md

Environment:
 Test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2300
 SDL build: [https://github.com/smartdevicelink/sdl_core/tree/develop] ( e137556 )
 ATF: develop branch

dboltovskyi, 2020-06-03:
@theresalech Please notice issue previously was closed by mistake and it's still actual for Core 6.1
ATF script and logs: [2020-06-02_20-52-50_1919.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4720437/2020-06-02_20-52-50_1919.tar.gz)

theresalech, 2020-06-03:
@dboltovskyi thank you for letting us know.  This issue has not yet been prioritized for the Core 6.2 release.  Please let us know if this is high priority for Luxoft in the next release, and if so, if you will be submitting a PR to resolve the issue.  Thanks!

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible

jacobkeeler, 2020-09-09:
Fixed by #3476

closed 2020-09-09 by jacobkeeler