smartdevicelink/sdl_core #1920: Multiple media and navigation apps activation during active embedded navigation+audio mixing supported

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Multiple media and navigation apps activation during active embedded navigation+audio mixing supported

## Preconditions
- "MixingAudioSupported" = true at .ini file
- SDL and HMI are started.
- Three apps running on system:
  a. media app_1 in FULL and AUDIBLE
  b. navigation app_2 in LIMITED and AUDIBLE
  c. non-media app_3 in BACKGROUND and NOT_AUDIBLE
- User activates embedded navigation and HMILevel of apps were changed to:
  a. media app_1 in LIMITED and AUDIBLE
  b. navigation app_2 in BACKGROUND and NOT_AUDIBLE

## Steps to reproduce
1. User activates media app_1
HMI -> SDL: SDL.ActivateApp ( <appID_1> )
SDL -> HMI: SDL.ActivateApp (SUCCESS)
2. SDL -> media app_1: OnHMIStatus (FULL, AUDIBLE)
Note: embedded navigation is still active
3. User activates navigation app_2
HMI -> SDL: OnEventChanged (EMBEDDED_NAVI, isActive:false)
HMI -> SDL: OnAppDeactivated ( <appID_1> )
HMI -> SDL: SDL.ActivateApp ( <appID_2 )

## Actual result
SDL does not set required HMILevel and audioStreamingState.

## Expected result
1. SDL -> media app_1 : OnHMIStatus (LIMITED, AUDIBLE)
2. SDL -> HMI: SDL.ActivateApp (SUCCESS)
3. SDL -> navigation app_2 : OnHMIStatus (FULL, AUDIBLE)

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to Medium. Issue is related to setting wrong HMI level and/or audioStreamingState for activated application in some specific cases. May affect correct permissions calculation for media activated application.

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1. This is a high priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

theresalech, 2020-05-21:
@GetmanetsIrina can you please advise if you are able to reproduce this issue in the latest Core release?

dboltovskyi, 2020-06-03:
@theresalech Please notice issue is still actual for Core 6.1
ATF script and logs: [2020-06-02_20-53-27_1920.tar.gz](https://github.com/smartdevicelink/sdl_core/files/4720441/2020-06-02_20-53-27_1920.tar.gz)

theresalech, 2020-06-03:
Hi @dboltovskyi, thank you for the additional information. Will Luxoft be submitting a PR to resolve this issue?

dboltovskyi, 2020-06-11:
@theresalech Please notice we're not planning to create the fix in scope of 6.2, though will try to do as much as possible