smartdevicelink/sdl_core #1921: Processing invalid PT after cutting off unknown_parameter or unknown_RPC

GetmanetsIrina, 2017-11-24:
**Occurrence: Always** 

# Description
Processing invalid PT after cutting off unknown_parameter or unknown_RPC

## Preconditions
- SDL and HMI are started.
- App is registered.

## Steps to reproduce
1. SDL received UpdatedPT with at least one <unknown_parameter> or <unknown_RPC>
and after cutting off <unknown_parameter> or <unknown_RPC> UpdatedPT is invalid

## Actual result

## Expected result
SDL must log the error internally and discard Policy Table Update

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to High. Updating policy with invalid PTU (which contains unknown RPC or parameter) may affect application permissions calculating process.

GetmanetsIrina, 2017-12-27:
Issue is reproduced on develop branch 61c37f3763b7f84dca8f42ffdaa705d454d1b711.

LuxoftAKutsan, 2017-12-27:
Issue is related to https://github.com/smartdevicelink/sdl_core/issues/1885

LuxoftAKutsan, 2017-12-27:
Fix is available https://github.com/smartdevicelink/sdl_core/pull/1972 

But fix is partial. It includes only cutting off <unknown_parameter>  And It does not cut's of  <unknown_RPC>. 

This is done because functionality of cutting of <unknown_RPC> is also absent in Sync 3.2v2 repo. To make code aligned it was not implemented in opensdl.

Missing functionality of cutting <unknown_RPC> is not critical and does not breaks any use cases. 
In case of Mobile application will send RPC with unknown FunctionID, it will be rejected during parsing, and policy login won't be involved. Locating of <unknown_RPC> in PolicyTable does not affect other RPCs in any use cases. 

Implementation of of cutting <unknown_RPC> is possible, but it should be done both in openSDL and in Sync3.2v2. Also this regression of this implementation might be rather big. 

@robinmk  please considering this points, should cutting <unknown_RPC> functionality be implemented or not?

robinmk, 2017-12-28:
@LuxoftAKutsan , I have the same understanding as well - that unknown RPCs will be rejected/ignored early on and hence it would not break policies. If there is any change in this understanding please inform us as we will need to address at the earliest.
Let's also wait for @mrapitis  to comment.

mrapitis, 2018-01-02:
@LuxoftAKutsan I agree with @robinmk comment as well -- the RPC should send a generic response back to the mobile if it is of an unknown type (as it does today) which should not impact policies.

LuxoftAKutsan, 2018-02-05:
@robinmk @mrapitis good, so please approve that cutting <unknown_RPC> functionality should not be implemented.

mrapitis, 2018-05-25:
@GetmanetsIrina are there any remaining tasks required here?  It looks like we still have three items unchecked in the expected delivery section.  Can you please check them if this was an oversight?

GetmanetsIrina, 2018-05-29:
@mrapitis , task status is updated, thanks! https://github.com/smartdevicelink/sdl_core/pull/1972 is mentioned above.

LuxoftAKutsan, 2018-05-29:
@mrapitis Sorry, the #2006 is the correct PR.

mrapitis, 2018-05-29:
@LuxoftAKutsan how to #2006 and #1972 differ? it looks like #1972 was already reviewed?

LuxoftAKutsan, 2018-05-29:
@mrapitis #2006 also reviewed, but not by zhimin.
I would say that #1972 is not ready and looks like a draft version. #2006 is the same approach but is more clear and finished. It is my fail that there was 2 PRs, #1972 should be closed. I will close it if you don't mind

PR #2006 (2018-09-21): Fix/filter invalid parametres and rpcs

Fixes #1921

This PR is ready for review.

### Risk
This PR makes no API changes.

### Testing Plan
Tested with ATF script attached to  #1921

### Summary
Added generation most of policy enums. Added to policy enums RPC's
Added filtering of received RPC and parameters in policy update.

Added bson ad custom target to avoid cmake warning
Added logger initialization in unit tests run for analyzing unit test failures without debugging.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2 [... cut]
PR #2627 (2018-09-21): Update to filter invalid parametres and rpcs
Fixes #1921 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Set up policy server with unknown parameters, hmi levels, and module types.
Connect an app and perform successful policy table update.

### Summary
Update PR to filter out remaining policy table types:

- HMILevel
- AppHMIType
- Priority
- RequestType
- RequestSubType
- moduleType
### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
PR #2624 (2018-09-24): 4.1LTS Patch: Add Policy Table Filtering
[Things to note: Pull Requests **must** fix an issue. Discussion about the feature / bug takes place in the issue, discussion of the implementation takes place in the PR. Please also see the [Contributing Guide](https://github.com/smartdevicelink/sdl_core/blob/master/.github/CONTRIBUTING.md) for information on branch naming and the CLA.

Delete the above section when you've read it.]

Fixes #1921 For 4.1 Release. Separate PR for 5.0 release.

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Perform a policy  [... cut]
closed 2018-09-21 by JackLivio