smartdevicelink/sdl_core #1925: SDL must start PTU for navi app right after app successfully registration

AStasiuk, 2017-11-24:
**Occurrence: Always** 

# Description
SDL must start PTU for navi app right after app successfully registration

## Preconditions
 - SDL and HMI are started.
 - App is registered.

## Steps to reproduce
1. In case
navigation app connects and sucessfully registers on SDL (opens RPC 7 service)
and PolicyTable has NO "certificate" at "module_config" section of LocalPolicyTable

## Actual result
N/A

## Expected result
SDL must:
start PolicyTableUpdate process on sending SDL.OnStatusUpdate(UPDATE_NEEDED) to HMI to get "certificate"

# Environment

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [x] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2017-12-01:
Priority was set to High. Issue makes impossible starting of secure sessions because of absent certificate in PT. TLS handshake will not be resumed until valid certificate will be received.

dboltovskyi, 2017-12-27:
Issue is reproduced on develop branch (0b19cf4)

jacobkeeler, 2018-01-31:
Fixed on current develop branch

PR #1971 (2018-01-31): Fix starting behavior of secured services for navi/non-navi applications
Fixes #1925, #1924, #1923, #1922, #1912, #1894, #1891, #1888

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
Existing unit tests will be updated according to new logic and new unit tests will be provided.

### Summary
There was a wrong logic around processing of `ForceProtectedService` and `ForceUnprotectedService`
params from ini file and sending start service ACK/NACK for secured services. All these test cases could be covered with one pull request. Also there was a wro [... cut]
closed 2018-01-31 by jacobkeeler