smartdevicelink/sdl_core #1940: Master Branch Build Fails with BUILD_TESTS=on

BrandonHe, 2017-12-06:
### Bug Report
[Summary]

##### Reproduction Steps
1. Check out latest master branch
2. Modify CMakeLists.txt to set BUILD_TESTS=on
3. create a build directory ($ mkdir build && cd build)
4. Give a build ($ sudo cmake ../ && make). Type your sudo password if asked

##### Expected Behavior
Build without any issue

##### Observed Behavior
```shell
[ 91%] Generating CXX unity source src/components/application_manager/test/commands/cotire/commands_test_CXX_unity.cxx
Scanning dependencies of target commands_test_unity
[ 91%] Building CXX object src/components/application_manager/test/commands/CMakeFiles/commands_test_unity.dir/cotire/commands_test_CXX_unity.cxx.o
virtual memory exhausted: Cannot allocate memory
make[2]: *** [src/components/application_manager/test/commands/CMakeFiles/commands_test_unity.dir/cotire/commands_test_CXX_unity.cxx.o] Error 1
make[1]: *** [src/components/application_manager/test/commands/CMakeFiles/commands_test_unity.dir/all] Error 2
make: *** [all] Error 2

```
##### OS & Version Information
* Ubuntu14.04 x86_64 Virtual Machine

LuxoftAKutsan, 2017-12-06:
@BrandonHe please specify amount of RAM on your Ubuntu14.04 x86_64 Virtual Machine

yangbiao01, 2017-12-06:
I met this problem in  Ubuntu16.04 x86_64 Virtual Machine too,The test is that memory has gone up until the virtual machine card is dead

BrandonHe, 2017-12-06:
@LuxoftAKutsan 
Thanks for your response. I created with 2048 MB size for RAM.

LuxoftAKutsan, 2017-12-06:
@BrandonHe we are using [Cotire](https://github.com/sakra/cotire) for default SLD build with UT. It means that commands unit tests are linked to one huge binary. I suppose this linkage require more than 2 GB of RAM. We have no strict requirements for the amount of ram to build SDL. 
You can either increase RAM bot you build station or build SDL without Cotire `cmake $PATH_TO_SRC -DBUILD_TESTS=ON -DUSE_COTIRE=OFF `
In that case all unit also builds but to multiple small binaries.

BrandonHe, 2017-12-11:
@LuxoftAKutsan 
I have tried with your suggestion that build SDL without Cotire. 
Still fails at the **commands** unit test:

```shell
07 [ 92%] Building CXX object src/components/application_manager/test/commands/CMakeFiles/commands_test.dir/mobile/dial_number_request_test.cc.o
308 [ 92%] Building CXX object src/components/application_manager/test/commands/CMakeFiles/commands_test.dir/mobile/get_dtcs_request_test.cc.o
309 /home/brandon/work/ford/smartdevicelink/sdl_core/src/components/application_manager/test/commands/mobile/get_dtcs_request_test.cc: In constructor ‘    test::components::commands_test::mobile_commands_test::get_dtcs_request::GetDTCsRequestTest::GetDTCsRequestTest()’:
310 /home/brandon/work/ford/smartdevicelink/sdl_core/src/components/application_manager/test/commands/mobile/get_dtcs_request_test.cc:73:38: error: ‘me    ssage_helper_mock_’ was not declared in this scope
311      Mock::VerifyAndClearExpectations(message_helper_mock_);
312                                       ^
313 make[2]: *** [src/components/application_manager/test/commands/CMakeFiles/commands_test.dir/mobile/get_dtcs_request_test.cc.o] Error 1
314 make[1]: *** [src/components/application_manager/test/commands/CMakeFiles/commands_test.dir/all] Error 2
315 make: *** [all] Error 2
```
For more details, please reference the attached build log file:
[make_test_cotire_off.log](https://github.com/smartdevicelink/sdl_core/files/1546336/make_test_cotire_off.log)

LuxoftAKutsan, 2017-12-11:
@BrandonHe  this issue fixed with PR https://github.com/smartdevicelink/sdl_core/pull/1934

BrandonHe, 2017-12-11:
@LuxoftAKutsan Thanks, I will test and close the issue after the PR is merged.

jacobkeeler, 2018-04-06:
Fix with the release of 4.5.0

closed 2018-04-06 by jacobkeeler