smartdevicelink/sdl_core #1943: Memory Corruption with DATA_SERVICE_ACK

VGunasekaranHarman, 2017-12-08:
### Bug Report
[Summary]
Hi, We are observing a memory corruption with SDL while sending the DATA_SERVICE_ACK for the mobile navigation scenario. If we comment the PostMessage in function ProtocolHandlerImpl::SendFramesNumber, then we are not observing any memory corruption and no crash is observed.
[TomTomCrashLog_8thDec_ToLivio.zip](https://github.com/smartdevicelink/sdl_core/files/1542695/TomTomCrashLog_8thDec_ToLivio.zip)

##### Reproduction Steps
1. [Step 1] HU is ON. Connect the Android phone having TomTom application
2. [Step 2] Once mobile navigation projection starts, please wait for some time.
3. [Step 3]

##### Expected Behavior
Mobile navigation projection should work continuously.

##### Observed Behavior
After sometime, there is a crash observed in SDL.

##### OS & Version Information
* OS/Version: Linux - iMX6 board
* SDL Core Version: SDL 4.2
* Testing Against: QT based HMI on HU and Tomtom app from mobile

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this issue   reproduce with the reproduction steps described? Also, Does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@VGunasekaranHarman, can you please advise if you are seeing this issue when using Core 6.1?

dboltovskyi, 2020-05-11:
@theresalech Please notice issue is not reproduced under described conditions using Core 6.1.
However we identified similar unexpected behavior related SDL logger and video streaming which is described it in https://github.com/smartdevicelink/sdl_core/issues/3358

theresalech, 2020-05-11:
Closing as not reproducible in the latest version of Core.  Issue 3358 can be raised to the Steering Committee as priority for the Core 6.2 release if desired.

closed 2020-05-11 by theresalech