smartdevicelink/sdl_core #1944: Unable to register application if VR in supported by Head unit

LuxoftAKutsan, 2017-12-08:
### Bug Report
In case if there is no VR support on Head unit, SDL returns WRONG_LANGUAGE result code to RegisterAppInterface response.  

##### Reproduction Steps
1. Remove VR related data from hmi_capabilities 
2. Register Application

##### Expected Behavior
Succesful registration

##### Observed Behavior
N/A

##### OS & Version Information
* OS/Version: SYNC
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/61c37f3763b7f84dca8f42ffdaa705d454d1b711

LuxoftAKutsan, 2017-12-08:
This is not the issue because SDL have defaults languages even if is it empty in hmi_capabilities.json
Even if there is no VR support on HU, SDL will by default setup EN-US language as active VR:
```
if (check_existing_json_member(root_json, "VR")) {
      Json::Value vr = root_json.get("VR", "");
          if (check_existing_json_member(vr, "language")) {
            const std::string lang = vr.get("language", "EN-US").asString();
            set_active_vr_language(MessageHelper::CommonLanguageFromString(lang));
          } else {
            set_active_vr_language(
                MessageHelper::CommonLanguageFromString("EN-US"));
          }

```
language, and response to mobile with SUCCESS result code on registration, with hmi_capabilities.

This sequence was successfully  tested with ATF script https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1757

jacobkeeler, 2020-06-15:
Not reproducible in develop, closing

closed 2020-06-15 by jacobkeeler