smartdevicelink/sdl_core #2009: Crash in Event Dispatcher

VGunasekaranHarman, 2018-02-07:
### Bug Report
Crash observed in Event Dispatcher

##### Reproduction Steps
1. Make the BT connection between HU and Android phone with an Applink app opened(Preferably, Wetter.De)
2. Open the app and keep perform the soft button press of two buttons alternatively.

##### Expected Behavior
There should be any crash

##### Observed Behavior
After sometime, there is a crash observed in SDL - Event Dispatcher.

##### OS & Version Information
* OS/Version: Linux iMX6 board
* SDL Core Version: SDL 4.2
* Testing Against: Used Wetter.De app

##### Test Case, Sample Code, and / or Example App

VGunasekaranHarman, 2018-02-07:
Hi,

Have attached the log taken during the scenario.  
[applinkLog6.zip](https://github.com/smartdevicelink/sdl_core/files/1703102/applinkLog6.zip)

We had taken the below code/patch from SDL 4.4 which fixes the issue 1737
---------------------------------------------------------
void EventDispatcherImpl::raise_event(const Event& event) {
  {
  ...
    // Call observers
  EventObserver* temp;
  while (!observers_.empty()) {
    observer_lock_.Acquire();
    temp = *observers_.begin();
    observers_.erase(observers_.begin());
    observer_lock_.Release();
    temp->on_event(event);
  }
}
---------------------------------------------------------

In the log, we are observing that after temp is assigned and lock is released, there is a context switch which removes all the entry in observers_ and so temp become invalid. Calling the function on_event on the invalid pointer makes the crash.

Is this addressed recently and fix is added??

Thanks.

VGunasekaranHarman, 2018-02-12:
Hi,
We figured out the root cause for this problem.

1. Mobile app is sending the RPCs with same correlation ID from same app.
2. The newer request is not added to the map in RequestController and RequestInfo (request_controller.cc and request_info.cc) if any older request is present with same correlation ID.
3. Even though the newer request is not added to the queue, the HMI RPC is sent to HMI for newer request.
4. Since the request is not added to map, it will get deleted once it sends HMI RPC to HMI and it goes out of scope.
5. When the response comes for the newer HMI RPC from HMI, then it tries to do raise event on the mobile RPC which got deleted in step 4.

Is there any fix added for the same or its planned??
Please correct me if I am wrong.

Thanks.

jacobkeeler, 2018-02-20:
I do not believe that the root cause you described is correct for this issue.

1. The issue you described relates to notifications (which have no correlation ID), not messages from the mobile app
2. I tested sending sets of 20 RPCs with the same correlation ID in quick succession. The logs showed several cases where the request was not added to the `waiting_for_response` queue (`Request with app_id = 65539; corr_id 200 Already exist`), but no crash ever occurred.
3. In the logs you provided in your previous comment, this `corr_id X already exist` message is never seen, meaning that a message with a duplicate correlation ID is never seen.

I am also a bit confused about your previous comment, as `temp` would not be invalidated after being set just because the observer list was cleared. Notably, temp has already been removed from the observers list before the lock is released. 

I do, however, see that there is an issue where the observer list could be emptied before the lock is acquired. This issue happens to be fixed in develop by #1950 (maybe this will fix the issue on your side as well).

One other note is that I was unable to reproduce the issue using the original steps that you provided.  Does the frequency with which you alternate pressing soft buttons make a difference, or does the crash just occur after doing it for long enough?

VGunasekaranHarman, 2018-02-21:
Hi @jacobkeeler,

Thanks for the input.

1. Yes, agreed that notification RPCs do not have correlation ID.
2. I was mentioning the mobile request RPCs which needs responses.
3. Ideally, the mobile RPC request's Run (Eg:show_request) will get executed which sends the HMI RPC to HMI. Once the response comes from HMI, it will raise the event to send response to the mobile RPC request.
4. Our deployment model is that there will be an Applink layer above SDL core where both Applink layer and SDL core will run in same process and HMI runs in separate process which just does rendering(That is, no RPC at all in HMI layer, It will be in only Applink layer)
5. In our case, the response to the HMI RPC(Eg: UI.Show) is coming immediately where it will raise event during the deletion of mobile RPC request observer and it causes crash.
6. We will try the patch that you have mentioned (#1950) and get back.
7. BTW, the shared log has the log "corr_id X Already exist". Below are snippets (Exactly, the problem faced with correlation ID 575)
--------------------------------------------------------------------------
[2523][2018:02:07 15:42:36:979]request_controller.cc:518 threadMain::Execute MobileRequest corr_id = 575 with timeout: 10000
[2523][2018:02:07 15:42:38:878]request_info.cc:125 Add::Add request app_id = 65543; corr_id = 575
[2523][2018:02:07 15:42:38:878]request_info.cc:143 Add::Request with app_id = 65543; corr_id 575 Already exist 
[[2523][2018:02:07 15:42:38:889]request_controller.cc:518 threadMain::Execute MobileRequest corr_id = 575 with timeout: 10000
[2523][2018:02:07 15:42:39:076]request_info.cc:125 Add::Add request app_id = 65543; corr_id = 575
[2523][2018:02:07 15:42:39:076]request_info.cc:143 Add::Request with app_id = 65543; corr_id 575 Already exist
[2523][2018:02:07 15:42:39:092]request_controller.cc:518 threadMain::Execute MobileRequest corr_id = 575 with timeout: 10000
[2523][2018:02:07 15:42:39:643]request_info.cc:125 Add::Add request app_id = 65543; corr_id = 575
[[2523]2524[2018:02:07 15:42:39:644]request_info.cc:143 Add::Request with app_id = 65543; corr_id 575 Already exist
--------------------------------------------------------------------------

Thanks,
Varadha.

jacobkeeler, 2018-02-21:
You are correct, the message is present, don't quite know how I missed that.

So, I'd have to guess that the app is sending `Show` messages in response to the `OnButtonPress` notifications for the soft buttons, and that is how this issue arose?

Regardless, it seem to me like the fix I sent you (#1950) should probably fix the issue, since it would prevent any other function from tampering with the observers while looping through them. My understanding is that it is likely crashing on this line:
```
EventObserver* temp = *observers_.begin();
```
when observers is empty, which shouldn't be possible with this fix.

VGunasekaranHarman, 2018-02-22:
Hi @jacobkeeler 

Thank you for the quick input.

Yes, the app is sending the Show request on getting OnButtonPress and there are many error responses sent from the SDL also for Show request due to the error "Reference Image not Found". That is, the image that UI.Show request is referring is not available.

We will check with the patch (#1950) and get back.

Thanks,
Varadha.

VGunasekaranHarman, 2018-02-22:
Hi @jacobkeeler 

We have verified the scenario with the fix mentioned (#1950).
We are still observing the crash.
However, we made below change in request_controller.cc and not observing the crash.

**From:**
void RequestController::Worker::threadMain() {
...
request_controller_->waiting_for_response_.Add(request_info_ptr);
...
}

**To:**
void RequestController::Worker::threadMain() {
...
    if (!request_controller_->waiting_for_response_.Add(request_info_ptr)) {
      LOG4CXX_ERROR(logger_, "Failed to Add Request - PROTOCOL_VIOLATION");
      continue;
    }
...
}

Thanks,
Varadha.

jacobkeeler, 2018-03-23:
This solution works for the most part, the only improvement I would include would be to actually send a `PROTOCOL_VIOLATION` response to the mobile app, this may be difficult given the current API, however.

jacobkeeler, 2018-05-16:
Closing with the merge of #2101

closed 2018-05-16 by jacobkeeler