smartdevicelink/sdl_core #2029: No PTU Exchange Trigger after x Kms

VGunasekaranHarman, 2018-02-27:
### Bug Report
No PTU Exchange Trigger after x Kms

##### Reproduction Steps
1. Register apps via BT medium from Android phone.
2. Simulate via CAN signal that it travelled more than x Kms
3. x Kms referred here is value of exchange_after_x_kilometers  in policy table.

##### Expected Behavior
PTU should be triggered

##### Observed Behavior
PTU is not triggered

##### OS & Version Information
* OS/Version: Linux iMX6
* SDL Core Version: SDL 4.2
* Testing Against: SyncProxyTester

##### Test Case, Sample Code, and / or Example App
[Paste a link to a PR, gist, or other code that exemplifies this behavior]

VGunasekaranHarman, 2018-02-27:
Hi,
We have taken the fix mentioned at https://github.com/smartdevicelink/sdl_core/pull/1129/files
Even after this, there is no PTU trigger happening after x Kms.

We feel that ODOMETER should be subscribed from SDL irrespective of its subscription from app. So that, on getting the ODOMETER value, SDL will calculate and trigger PTU with the fix - https://github.com/smartdevicelink/sdl_core/pull/1129/files

Please share your opinion on this.

Thanks.

JackLivio, 2018-03-01:
@VGunasekaranHarman Please [check this value exists in your policy tables](https://github.com/smartdevicelink/sdl_core/blob/master/src/appMain/sdl_preloaded_pt.json#L6)

A PTU will be triggered after every multiple of that number. Your head unit must send sdl core an OnVehicleData message for Core to know what the odometer value is

```
{"jsonrpc":"2.0","method":"VehicleInfo.OnVehicleData","params":{"odometer":1300005}}
```

This functionality has been tested and verified on 4.4. (Master branch). Please consider upgrading your version of sdl core to a newer release.

closed 2018-03-01 by JackLivio