smartdevicelink/sdl_core #2065: [SDL 0141] Supporting simultaneous multiple transports

theresalech, 2018-03-16:
Proposal: [Supporting simultaneous multiple transports](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0141-multiple-transports.md)

> This proposal aims to support multiple transports between Core and Proxy.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/405

Steering Committee Decision:

>The Steering Committee has voted to accept this proposal with revisions. The revisions are as follows:

- Introduce a new Control Frame, `RegisterSecondaryTransport`, which will be used once a Secondary Transport is connected to notify Core that the Proxy has established a connection and tell Core which session is registering.
- Rename `Transport Config Update` to `TransportEventUpdate`. The original parameters will be refactored to a flat structure.
- StartServiceACK for the RPC service will contain a Secondary Transport param that will contain an array of potential secondary transports instead of a single value. The service to transport priority parameters will become flat instead of document style; each service param will contain an ordered array in terms of priority of potential transports.
- Proxy connection retry strategy will include the new `TransportEventUpdate` Control Frame. When a connection is interrupted/ended, a `TransportEventUpdate` control frame will be sent to the proxy on the primary transport. If the proxy is to retry (in case of WiFi), it will include the IP address and Port. If the proxy is not to retry, the frame will contain either empty strings for IP address and port or not include the parameters at all.
- Will recommend, but not require, using 5GHz
- Proxy implementations will be discussed outside of this proposal.

>The proposal .md was updated on March 16, 2018 to include the agreed upon revisions.

theresalech, 2018-04-08:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [Accepted multiple transport proposal: Addition of ini entry and amending header of Register Secondary Transport NAK](https://github.com/smartdevicelink/sdl_evolution/pull/434).

jacobkeeler, 2018-07-06:
Closing with the merge of #2164

closed 2018-07-06 by jacobkeeler

zcloud-jzy, 2019-03-26:
This feature is still very good, continue to study in depth.

PR #2164 (2018-07-06): Feat/mt transport changes
Fixes #2065, #2130 
This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
All new functionality has been added to the unit test suite.

### Summary
This PR implements the features described in the issues:
•	SDL 0141 - Supporting simultaneous multiple transports
•	SDL 0149 - Add capability to disable resumption based on app type and transport type

### Changelog
##### Breaking Changes
* No breaking changes

### CLA
- [X] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegP [... cut]