smartdevicelink/sdl_core #2071: SDL doesn't consent mobile device

dboltovskyi, 2018-03-19:
**Occurrence: Always** 

# Description
SDL doesn't consent mobile device

## Preconditions
1. SDL is built with EXTERNAL_PROPRIETARY flag
2. SDL and HMI are started
3. App is registered

## Steps to reproduce
1. Activate app with the sequence:
```
HMI->SDL: SDL.ActivateApp
SDL->HMI: SUCCESS: SDL.ActivateApp
HMI->SDL: SDL.OnAllowSDLFunctionality(allowed = true, device = { <id>, <name> })
```

## Expected result
```
SDL->HMI: BC.ActivateApp
HMI->SDL: SUCCESS: BC.ActivateApp
SDL->App: OnHMIStatus(FULL)
```

## Actual result
```BC.ActivateApp``` request is not sent

## Test script
All scripts in:
* [External_UCS/External_Consent_OFF](https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop/test_scripts/Policies/External_UCS/External_Consent_OFF) 
* [External_UCS/External_Consent_ON](https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop/test_scripts/Policies/External_UCS/External_Consent_ON)

Many other scripts related to EXTERNAL_PROPRIETARY flow

# Environment
SDL: https://github.com/smartdevicelink/sdl_core/tree/develop (73f3298)
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop (ebf5c94)
ATF Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop (c2b2037)
OS: Ubuntu 16.04.3 LTS x64

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2018-03-19:
Please notice issue is not reproduced at [4.3.0RC](https://github.com/smartdevicelink/sdl_core/tree/4.3.0RC)

LuxoftAKutsan, 2018-03-22:
Bug is both in scripts and in SDL.
Scripts fix : https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1857

LuxoftAKutsan, 2018-03-22:
Bug is both in scripts and in SDL.
Scripts fix : https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1857 
SDL fix : #2085

theresalech, 2018-03-26:
Closed via #2085

closed 2018-03-26 by theresalech