smartdevicelink/sdl_core #2072: SDL doesn't send OnPermissionsChange in case of external user consent

dboltovskyi, 2018-03-19:
**Occurrence: Always** 

# Description
SDL doesn't send OnPermissionsChange in case of external user consent

## Preconditions
1. SDL is built with EXTERNAL_PROPRIETARY flag
2. There functional group and appropriate entity are configured which requires external user consent
3. SDL and HMI are started
4. App is registered

## Steps to reproduce
1. Make external user consent:
```
HMI->SDL: SDL.OnAppPermissionConsent(externalConsentStatus = {
{<entityType>, <entityID>, <status>}
})
```

## Expected result
```
SDL->App: OnPermissionsChange
```

## Actual result
OnPermissionsChange notification is not sent

## Test script
[001](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/External_UCS/External_Consent_OFF/001_ATF_Policies_External_Consent_OFF_disallowed_by_external_consent_entities_on_data_consent_disallowed.lua)
[004](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/External_UCS/External_Consent_OFF/004_ATF_Policies_External_Consent_OFF_disallowed_by_external_consent_entities_off_data_consent_disallowed.lua)
[007](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/External_UCS/External_Consent_OFF/007_ATF_Policies_External_Consent_OFF_disallowed_by_external_consent_entities_omitted_data_consent_disallowed.lua)

[001](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/External_UCS/External_Consent_ON/001_ATF_Policies_External_Consent_ON_disallowed_by_external_consent_entities_off_data_consent_disallowed.lua)
[004](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/External_UCS/External_Consent_ON/004_ATF_Policies_External_Consent_ON_disallowed_by_external_consent_entities_on_data_consent_disallowed.lua)
[007](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/External_UCS/External_Consent_ON/007_ATF_Policies_External_Consent_ON_disallowed_by_external_consent_entities_omitted_data_consent_disallowed.lua)
[031](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/External_UCS/External_Consent_ON/031_ATF_Policies_External_Consent_ON_externalConsentStatus_is_not_sent.lua)

# Environment
SDL: https://github.com/smartdevicelink/sdl_core (6212599)
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop (ebf5c94)
ATF Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop (c2b2037)
OS: Ubuntu 16.04.3 LTS x64

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

dboltovskyi, 2018-03-19:
Please notice issue is not reproduced at [4.3.0RC](https://github.com/smartdevicelink/sdl_core/tree/4.3.0RC)

LuxoftAKutsan, 2018-03-23:
Fix available https://github.com/smartdevicelink/sdl_core/pull/2086
Also depends on fix   https://github.com/smartdevicelink/sdl_core/pull/2085
Also depends on scripts fix https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1857

theresalech, 2018-03-29:
Closed via #2086

closed 2018-03-29 by theresalech