smartdevicelink/sdl_core #2073: Build fails on `develop` branch when ENABLE_SECURITY=OFF

shoamano83, 2018-03-20:
### Bug Report
On latest `develop` branch (commit 73f3298cf7), build fails if `ENABLE_SECURITY` is configured to OFF.

##### Reproduction Steps
1. Edit CMakeLists.txt to disable ENABLE_SECURITY flag
2. Give a build (create a dir then cmake ../sdl_core && make)

##### Expected Behavior
Build succeeds.

##### Observed Behavior
Build fails with following error message:
```
[ 19%] Building CXX object src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/protocol_handler_impl.cc.o
/home/samano/projects/sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc: In constructor 'protocol_handler::ProtocolHandlerImpl::ProtocolHandlerImpl(const protocol_handler::ProtocolHandlerSettings&, protocol_handler::SessionObserver&, connection_handler::ConnectionHandler&, transport_manager::TransportManager&)':
/home/samano/projects/sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:80:5: error: class 'protocol_handler::ProtocolHandlerImpl' does not have any field named 'is_ptu_triggered_'
     is_ptu_triggered_(false)
     ^
/home/samano/projects/sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc: In member function 'protocol_handler::RESULT_CODE protocol_handler::ProtocolHandlerImpl::HandleControlMessageStartSession(ProtocolFramePtr)':
/home/samano/projects/sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:1411:17: error: unused variable 'protocol_version' [-Werror=unused-variable]
   const uint8_t protocol_version = packet->protocol_version();
                 ^
cc1plus: all warnings being treated as errors
src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/build.make:86: recipe for target 'src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/protocol_handler_impl.cc.o' failed
make[2]: *** [src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/protocol_handler_impl.cc.o] Error 1
CMakeFiles/Makefile2:1216: recipe for target 'src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/all' failed
make[1]: *** [src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

##### OS & Version Information
* OS version: Ubuntu 16.04 LTS (64bit)
* SDL Core Version: latest `develop` branch, commit 73f3298cf7

##### Test Case, Sample Code, and / or Example App
N/A

shoamano83, 2018-03-20:
I also noticed that release/4.5.0 branch has same issue. Here is the error log for the branch.
```
Scanning dependencies of target ProtocolHandler
[ 20%] Building CXX object src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/protocol_packet.cc.o
[ 20%] Building CXX object src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/protocol_handler_impl.cc.o
/home/samano/projects/sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc: In constructor 'protocol_handler::ProtocolHandlerImpl::ProtocolHandlerImpl(const protocol_handler::ProtocolHandlerSettings&, protocol_handler::SessionObserver&, connection_handler::ConnectionHandler&, transport_manager::TransportManager&)':
/home/samano/projects/sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:80:5: error: class 'protocol_handler::ProtocolHandlerImpl' does not have any field named 'is_ptu_triggered_'
     is_ptu_triggered_(false)
     ^
/home/samano/projects/sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc: In member function 'protocol_handler::RESULT_CODE protocol_handler::ProtocolHandlerImpl::HandleControlMessageStartSession(ProtocolFramePtr)':
/home/samano/projects/sdl_core/sdl_core/src/components/protocol_handler/src/protocol_handler_impl.cc:1411:17: error: unused variable 'protocol_version' [-Werror=unused-variable]
   const uint8_t protocol_version = packet->protocol_version();
                 ^
cc1plus: all warnings being treated as errors
src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/build.make:86: recipe for target 'src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/protocol_handler_impl.cc.o' failed
make[2]: *** [src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/src/protocol_handler_impl.cc.o] Error 1
CMakeFiles/Makefile2:1216: recipe for target 'src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/all' failed
make[1]: *** [src/components/protocol_handler/CMakeFiles/ProtocolHandler.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```
I have updated the title accordingly.

jacobkeeler, 2018-03-22:
Closing with the merge of #2076

closed 2018-03-22 by jacobkeeler

lb75, 2018-07-25:
Build fails with the same error message. release/4.5.1

PR #2076 (2018-03-22): Fix build issues and test failures when building with ENABLE_SECURITY=OFF

Fixes #2073 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Fix build issues with ENABLE_SECURITY=OFF with the introduction of the HandshakeHandler in #1971, as well as any test failures resulting from these changes.

### Changelog

##### Bug Fixes
* Fix build and test failures when ENABLE_SECURITY=OFF

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)