smartdevicelink/sdl_core #2087: SDL accidentally crashed during application registration

dboltovskyi, 2018-03-23:
**Occurrence: Sometimes (7/1000)** 

# Description
SDL accidentally crashed during application registration

## Steps to reproduce
There are no exact steps
Please find a few cases in attachment
(Core dump included)

## Expected result
No crash

## Actual result
Crash

## Test script
[1 - API/017_Slider_PositiveCase_SUCCESS](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/API/017_Slider_PositiveCase_SUCCESS.lua)
[2 - Resumption/002_Resumption_app_registered_in_more_than_30sec_after_BC.OnReady](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/Resumption/002_Resumption_app_registered_in_more_than_30sec_after_BC.OnReady.lua)
[3 - Resumption/003_Resumption_App_Unregister_itself](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/Resumption/003_Resumption_App_Unregister_itself.lua)
[4 - API/016_ShowConstantTBT_PositiveCase_SUCCESS](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/API/016_ShowConstantTBT_PositiveCase_SUCCESS.lua)
[5 - Resumption/002_Resumption_app_registered_in_more_than_30sec_after_BC.OnReady](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/Resumption/002_Resumption_app_registered_in_more_than_30sec_after_BC.OnReady.lua)
[6 - API/005_AddSubMenu_PositiveCase_SUCCESS](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/API/005_AddSubMenu_PositiveCase_SUCCESS.lua)
[7 - API/041_Speak_Non_Media_PositiveCase_SUCCESS](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Smoke/API/041_Speak_Non_Media_PositiveCase_SUCCESS.lua)

# Environment
SDL: https://github.com/smartdevicelink/sdl_core (6212599)
ATF: https://github.com/smartdevicelink/sdl_atf/tree/develop (ebf5c94)
ATF Scripts: https://github.com/smartdevicelink/sdl_atf_test_scripts/tree/develop (c2b2037)
OS: Ubuntu 16.04.3 LTS x64

# Attachments

# Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ] **UTs add/update _(not required)_**
- [ ] ATF tests add/update
- [x] **Manual tests _(not required)_**
- [x] **Add/update CI plans/jobs _(not required)_**
- [x] **SDD updates _(not required)_**
- [x] **Guidelines update ([sdl_core_guides](https://github.com/smartdevicelink/sdl_core_guides)) _(not required)_**
- [x] **Guidelines update ([sdl_hmi_integration_guidelines](https://github.com/smartdevicelink/sdl_hmi_integration_guidelines)) _(not required)_**

E-SAITO-TMC, 2020-04-17:
@joeygrover -san @theresalech -san This issue is a top priority for Toyota. Does this issue reproduce with when run test scripts? Also, does this issue occur in the latest version?

theresalech, 2020-04-20:
Hi @E-SAITO-TMC, we have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.
@dboltovskyi, can you please advise if the Luxoft team is seeing this issue in Core 6.1?

dboltovskyi, 2020-05-11:
@theresalech Please notice this issue is not actual for Core 6.1.

theresalech, 2020-05-11:
Closing as not reproducible in latest version of Core.

closed 2020-05-11 by theresalech