smartdevicelink/sdl_core #2106: Stopping SDL Core by using SIGTERM causing SIGSEGV ( with Boost Integration)

madhuy, 2018-04-03:
##### Reproduction Steps
1. strat SDlcore
2. stop by using SIGTERM( kill -15 pid of sdlcore)

##### Expected Behavior
SDL core exits successfully with out SIGSEGV

##### Observed Behavior
observed SIGSEGV

##### OS & Version Information
* OS/Version: Ubuntu 16.04 LTS
* SDL Core Version:  latest from Develop branch
* Testing Against: HMI

Attaching GDB out put and smartdevicelinkcore logs 
[SmartDeviceLinkCore.log](https://github.com/smartdevicelink/sdl_core/files/1869784/SmartDeviceLinkCore.log)
[gdb.txt](https://github.com/smartdevicelink/sdl_core/files/1869785/gdb.txt)

madhuy, 2018-04-03:
Here is backtrace
```Thread 1 (Thread 0x7ffff7fb2740 (LWP 31624)):
#0  0x000000000215015b in std::__atomic_base<bool>::store (__m=std::memory_order_seq_cst, __i=true, this=0xc80)
    at /usr/include/c++/5/bits/atomic_base.h:374
#1  std::__atomic_base<bool>::operator= (this=0xc80, __i=true) at /usr/include/c++/5/bits/atomic_base.h:267
#2  0x0000000002149c3d in std::atomic<bool>::operator= (this=0xc80, __i=true) at /usr/include/c++/5/atomic:74
#3  0x00000000021702c7 in hmi_message_handler::WebsocketSession::Shutdown (this=0x0)
    at /home/madhu/git/github/sdl_core/src/components/hmi_message_handler/src/websocket_session.cc:32
#4  0x0000000002146586 in hmi_message_handler::CMessageBrokerController::exitReceivingThread (this=0x2a83f40)
    at /home/madhu/git/github/sdl_core/src/components/hmi_message_handler/src/mb_controller.cc:195
#5  0x00000000017873f0 in main_namespace::LifeCycle::StopComponents (this=0x7fffffffce10)
    at /home/madhu/git/github/sdl_core/src/appMain/life_cycle.cc:334
#6  0x000000000178e79b in main (argc=2, argv=0x7fffffffded8) at /home/madhu/git/github/sdl_core/src/appMain/main.cc:173```

jacobkeeler, 2018-04-03:
@madhuy Appears to be fixed in the `release/4.5.0` branch (originally I thought by #2033, but that appears to be in develop). Merging this branch into develop (which I've just done) should fix the issue there as well. Can you verify that this fixes the issue on your side (and close this issue if so)?

madhuy, 2018-04-03:
@jacobkeeler I tried with `develop` branch it is not crashing. Thanks for fixing. 
and also I have question regarding threads , is there any reason to create threads by using c++11 `std::thread` instead of `posix thread( createThread())` ? 
we have issues in certain platforms with std::thread.

conlain-k, 2018-07-25:
Occurs for me with `develop` on Ubuntu 18.04, same exact issue as before

iCollin, 2021-03-31:
Closed via #2420

closed 2021-03-31 by iCollin