smartdevicelink/sdl_core #2109: Invalid PTU results in infinite update loop

jacobkeeler, 2018-04-04:
### Bug Report
SDL doesn't respect the normal PTU retry sequence when it receives an invalid policy table from the update server, resulting in an infinite update loop. 

##### Reproduction Steps
1. Build SDL Core with default flags
2. Start Core
3. Perform a policy update with an invalid policy table from the cloud

##### Expected Behavior
SDL Core rejects the policy table and attempts an update after the specified retry timeout

##### Observed Behavior
SDL Core rejects the policy table and immediately attempts another policy table update, resulting in an endless PTU loop (assuming that the server continues sending an invalid policy table).

##### OS & Version Information
* OS/Version: Ubuntu 14.04
* SDL Core Version: https://github.com/smartdevicelink/sdl_core/commit/6212599da217e95899d63d4b33088082489841bd
* Testing Against: N/A

theresalech, 2018-04-12:
Fixed via #2114

jacobkeeler, 2018-04-16:
Reopening with new case on EXTERNAL_PROPRIETARY mode:

##### Reproduction Steps
1. Build SDL Core with default flags
2. Start Core
3. Perform a policy update with an invalid policy table from the cloud
4. When a retry is attempted after the failure, return a valid policy table from the server

Infinite PTU occurs (caused by the update status never being updated from UPDATE_NEEDED to UPDATING)

theresalech, 2018-04-17:
Closed via #2141

PR #2141 (2018-04-16): Fix infinite PTU when invalid PT, then a valid PT is received
Fixes #2109

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
Fixes case where an infinite PTU occurs after an invalid PT is received from the server, but a valid PT is received on retry.

### Changelog
##### Bug Fixes
* Fixes case where an infinite PTU occurs after an invalid PT is received from the server, but a valid PT is received on retry

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-04-17 by theresalech