smartdevicelink/sdl_core #2116: onButtonPress notification is not sent to LIMITED state app

t-yoshii, 2018-04-06:
### Bug Report
onButtonPress notification is not sent to LIMITED state app

##### Reproduction Steps
1. App subscribes button
2. Set app HMI level to LIMITED
3. Send onButtonPress event from HMI to Core

##### Expected Behavior
onButtonPress event is sent to mobile

##### Observed Behavior
onButtonPress event is not sent to mobile

##### OS & Version Information
* OS/Version: iOS 5.1.1 / Android
* SDL Core Version: 4.4, 4.5
* Testing Against: Our internal HMI devboard and internal test mobile apps

##### Test Case, Sample Code, and / or Example App
I will create PR

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix has been proposed for this bug. When will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is important for Toyota. Therefore, we would appreciate it if you can make this a high-priority.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-05-20:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3378

PR #3378 (2020-05-20): Forward OnButtonPressNotifications where HMI specifies appID to apps in LIMITED HMI Level
Fixes #2116

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Manual testing

### Summary
Initialize app in `OnButtonPressNotification::Run`

##### Bug Fixes
* If in an OnButtonPressNotification the HMI specifies an appID corresponding to an app in HMI Level Limited, Core will now forward that notification to the app.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-05-20 by iCollin