smartdevicelink/sdl_core #2129: [SDL 0150] Enhancing onHMIStatus with a New Parameter for Video Streaming State

theresalech, 2018-04-11:
Proposal: [Enhancing onHMIStatus with a New Parameter for Video Streaming State](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0150-video-streaming-state.md)

> This proposal adds a new parameter `videoStreamingState` to `onHMIStatus` notification so that SDL core can notify an app to stop video streaming. This helps SDL Core handle the use cases of the coexistence of a navigation app and a projection app or multiple navigation/projection apps.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/429

yang1070, 2018-05-22:
PR is here https://github.com/smartdevicelink/sdl_core/pull/2155

theresalech, 2018-06-13:
Closed via #2155

PR #2155 (2018-06-13): Feature/mobile projection 2
Fixes #2129

This PR is **ready** for review.

### Risk
This PR makes **major** API changes.

### Testing Plan
Tested with ATF : https://github.com/smartdevicelink/sdl_atf/tree/develop
ATF Scripts are "PASSED" from https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1879.

Manual testing:
Used HMI from: https://github.com/smartdevicelink/sdl_hmi/pull/53

### Summary
Implementation of proposal https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0150-video-streaming-state.md 

### Changelog
##### Breaking Changes

##### Enhancements

#####  [... cut]
PR #2424 (2018-08-06): make videostreamingstate optional
Fixes #2129

This PR is **ready** for review.

### Risk
This PR makes **minor** API changes.

### Testing Plan
Unit and ATF smoke tests pass
### Summary
Makes the videoStreamingState parameter optional in accordance with https://github.com/smartdevicelink/sdl_evolution/pull/548

### Changelog
##### Breaking Changes
* videoStreamingState is now optional

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2018-06-13 by theresalech