smartdevicelink/sdl_core #2134: [SDL 0119] SDL Passenger Mode

theresalech, 2018-04-12:
Proposal: [SDL Passenger Mode](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0119-SDL-passenger-mode.md)

> Allow a passenger to dismiss the lock screen to improve usability of applications. This re-visited proposal now includes a way for individual OEMs to control this mode via policy updates.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/359

Steering Committee Decision:

>The Steering Committee voted to accept this proposal with the following revisions:

- lockScreenDismissalEnabled param should not be mandatory
- Swipe method to dismiss the lock screen
- Be able to re-engage the lock screen if a notification is received, instead of allowing a lock screen to be disabled for the remainder of an app's session

>The proposal .md was updated on April 12, 2018 to include the agreed upon revisions.

FStoikov, 2019-02-15:
Jira ticket created for current case: https://adc.luxoft.com/jira/browse/FORDTCN-2016

theresalech, 2019-06-26:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0119 Revisions - SDL-passenger-mode](https://github.com/smartdevicelink/sdl_evolution/issues/769).  Please reference issue and markdown file for more details.

theresalech, 2019-07-17:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0119 Revisions - SDL Passenger Mode](https://github.com/smartdevicelink/sdl_evolution/issues/778).  Please reference issue and markdown file for more details.

theresalech, 2019-07-24:
Closed via #2783

closed 2019-07-24 by theresalech