smartdevicelink/sdl_core #2142: V5 Protocol Messages do not encrypt payload of Control Frame Message Types

mrapitis, 2018-04-16:
### Bug Report
Failure occurs when attempting to use a protected video or audio service with sdl core when using a max protocol version of 5 in the SmartDeviceLink.ini file.  This is because control frame message message types are being sent by sdl core with an encryption flag set to true, however the payload data is not encrypted causing failures when attempting to decrypt the payload data.  Downgrading to version 4 of the protocol by setting the max value to 4 works around the problem.

##### Reproduction Steps
1. Attempt to start a protected service with sdl core such as video or audio
2. Take note of the unencrypted payload data that is sent to the mobile when control frames are being used.
3. Take note that the protected service is not successfully established

##### Expected Behavior
When the encryption flag is set to true in a protocol message, the payload of the message is properly encrypted.

##### Observed Behavior
Unencrypted payload data for control frame type protocol messages even though the encryption bit is set to true in the message header.

##### OS & Version Information
* OS/Version: Ubuntu 16.04
* SDL Core Version: 4.5, 4.4
* Testing Against: VM running sdl with webHMI

jacobkeeler, 2018-04-23:
It appears as if it has been the standard to have control frames unencrypted based on [this line](https://github.com/smartdevicelink/sdl_core/blob/34e7256493ff0e6594029b9857d7e2aa31f5dbeb/src/components/protocol_handler/src/protocol_handler_impl.cc#L1201), which has been present in the project since it was added to GitHub. Is this actually the incorrect behavior?

mrapitis, 2018-04-23:
@AByzhynar can you please review and provide input based on the last comment? The control service type is used during the tls handshake process initially requiring unencrypted initialization frames, but why would this be the case for control frame types? For example a protocol message that has a video service type and a control frame type should allow for an encrypted payload.

GetmanetsIrina, 2018-05-25:
@mrapitis , @dboltovskyi will perform the investigation.

mrapitis, 2018-05-25:
@GetmanetsIrina @dboltovskyi during your investigation, can you please review the preliminary changes on the branch including potential fixes below and confirm if they are in line with your findings:

https://github.com/mrapitis/sdl_core/commit/7c382aeefd587d853b075d84c6dc88dfe88c5635

AStasiuk, 2018-06-04:
@mrapitis, review is done by @AKalinich-Luxoft

AKalinich-Luxoft, 2018-06-04:
@mrapitis
I've analysed your preliminary changes and suggest my own implementation for fixing this issue.
Fix will be retested in a separate integration branch within DTLS + GST features:
https://github.com/smartdevicelink/sdl_core/tree/fix/fix_ssl_handshake_flow_integration_branch_new_v5

PR #2235 (2018-06-18): Fix V5 protocol message bson params
Fixes #2142 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Covered by unit tests
ATF scripts

### Summary
There was a problem with V5 control frames encryption/decryption related to the wrong conditions in encrypt/decrypt functions in protocol manager. Also there was a problem with allocated bson params object passed as a pointer to another place.

NOTE
This fix should be merged after #2105 and #2068 features as it contains commit history from these branches. Please review only the last commit from this PR.

 [... cut]
closed 2018-06-18 by JackLivio