smartdevicelink/sdl_core #2143: Possible cyclic reference of shared pointers of `Application` class

shoamano83, 2018-04-17:
### Bug Report

There is a possible cyclic reference of shared pointers of `Application` class. Because of it, `ApplicationImpl` instance may not be released properly.

Looking at latest commit 10e17225d0a961f18493acd5ff8c1721e441cb14 of `develop` branch:

- An `Application` instance has `ApplicationState` in its member (https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/include/application_manager/application.h#L890)
- `ApplicationState` holds a vector of shared pointers of `HmiState` (https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/include/application_manager/application_state.h#L152)
- `HmiState` has a shared pointer of `Application` in its member (https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/include/application_manager/hmi_state.h#L203)

##### Reproduction Steps

Here are the repro steps to confirm that `ApplicationImpl` instance may not be released.

1. Check out latest `develop` branch (commit 10e17225d0a961f18493acd5ff8c1721e441cb14 )
2. Apply attached patch to add extra log output [0001-Add-printf-in-ApplicationImpl-ctor-and-dtor.patch.txt](https://github.com/smartdevicelink/sdl_core/files/1918359/0001-Add-printf-in-ApplicationImpl-ctor-and-dtor.patch.txt)
3. Update CMakeLists.txt file to enable `BUILD_TESTS` option
4. Give a build
5. Under the build directory, go to src/components/application_manager/test/
6. Run `./application_manager_test --gtest_filter=ApplicationImplTest.AddHmiState_GetCurrentState`

##### Expected Behavior

During the test, following logs should be observed:
```
>>> ApplicationImpl class constructor
>>> ApplicationImpl class destructor
```

##### Observed Behavior

Only `>>> ApplicationImpl class constructor` is shown. Please refer to attached log.
[application_impl_unit_test_output.txt](https://github.com/smartdevicelink/sdl_core/files/1918360/application_impl_unit_test_output.txt)

##### OS & Version Information

* OS/Version: Ubuntu 16.04 (64-bit)
* SDL Core Version: latest `develop` branch, commit 10e17225d0a961f18493acd5ff8c1721e441cb14 

##### Test Case, Sample Code, and / or Example App

If I apply attached change 
[0002-Do-not-merge-Change-shared-ptr-of-Application-to-nor.patch.txt](https://github.com/smartdevicelink/sdl_core/files/1918361/0002-Do-not-merge-Change-shared-ptr-of-Application-to-nor.patch.txt) and run the test again, I see both `>>> ApplicationImpl class constructor` and `>>> ApplicationImpl class destructor` logs. This is why I thought that cyclic reference is causing memory leak issue.

shoamano83, 2018-05-30:
It looks like the shared pointer in `HmiState` class has been added by commit a6791129fb1cfcd5dad96e4987b1c2664aafb61e, which was after 4.4 release.

I tried the repro steps with `4.4.0` tag and did not reproduce the issue. (i.e. both ">>> ApplicationImpl class constructor" and ">>> ApplicationImpl class destructor" messages were shown.)

jacobkeeler, 2019-03-04:
Closing with the merge of #2225

closed 2019-03-04 by jacobkeeler