smartdevicelink/sdl_core #2154: [SDL 0122] Handling VR help requests when application does not send VR help prompt

theresalech, 2018-04-23:
Proposal: [Handling VR help requests when application does not send VR help prompt](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0122-New_rules_for_providing_VRHelpItems_VRHelpTitle.md)

> When the user asks for "help" regarding an application, the help prompt that is played and the help menu that is displayed for that application is based on the information provided by the application through the [SetGlobalProperties RPC](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/interfaces/MOBILE_API.xml#L3125).
Through this proposal we propose a mechanism for handling "help" requests by the user when the application does not provide any such information.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/364

Steering Committee Decision:

> The Steering Committee has voted to accept this proposal with revisions.  The revision will be including a limit on `addCommand`s of 30.  It was also understood in the discussion that `CreateInteractionChoiceSet` should not trigger `SetGlobalProperties`.

>The proposal .md was updated on April 23, 2018 to include the agreed upon revisions.

theresalech, 2018-05-10:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0122 Revisions - Conditions for using the default values for helpPrompt and vrHelp parameters and update the default value for vrHelp in ini file](https://github.com/smartdevicelink/sdl_evolution/pull/469).

theresalech, 2018-06-28:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0122 Revisions - Remove 10 second timer from SDL-0122](https://github.com/smartdevicelink/sdl_evolution/pull/519).

jacobkeeler, 2018-07-16:
Closing with the merge of #2199

theresalech, 2018-08-29:
Proposal markdown file has been updated per the revisions included in accepted Evolution Proposal [SDL 0122 Revisions - Revise SDL-0122 to exclude synonyms](https://github.com/smartdevicelink/sdl_evolution/pull/589).

theresalech, 2018-09-04:
Closed via #2557

PR #2199 (2018-07-16): Feature/handling vr help requests
Fixes #2154

This PR is **ready** for review.

### Risk
This PR makes small API changes.

### Testing Plan
Unit tests are provided
ATF test scripts are provided

### Summary
Implemented Handling VR help requests when application does not send VR help prompt
PR #2186 (2018-07-16): Feature/help promt manager
Fixes #2154
This PR is **ready** for review.
### Risk
This PR makes small API changes.
### Testing Plan
Unit tests are provided
ATF test scripts are provided
### Summary
Implemented Handling VR help requests when application does not send VR help prompt
closed 2018-09-04 by theresalech