smartdevicelink/sdl_core #2165: iOS app may receive another HMI level when SDL.ActivateApp is issued right after BC.OnAppRegistered notification

shoamano83, 2018-05-01:
### Bug Report
If HMI issues SDL.ActivateApp right after it receives BC.OnAppRegistered notification, the HMI Level of the app may change to another one, for example, the default HMI level defined in policy table.

This is a timing issue. It doesn't always reproduce, and it may not reproduce if the head unit has powerful CPU.

##### Reproduction Steps
1. Terminate iOS app on the phone (by swiping it out)
2. Launch the app from head unit by using iAP AppLaunch command (note: this is achieved by head unit's proprietary implementation)
3. HMI issues SDL.ActivateApp as soon as the app is registered.

##### Expected Behavior
The app launches on the phone, gets registered to Core, and should get HMI level FULL with SDL.ActivateApp.

##### Observed Behavior
The app is launched, registered and got HMI level FULL. However, it will receive another HMI level (for example, the level defined in the policy table which is NONE in our case.)

##### OS & Version Information
* OS/Version: iPhone with iOS 11.1.
* SDL Core Version: release 4.3
* Testing Against: proprietary iOS app and proprietary HMI

##### Test Case, Sample Code, and / or Example App
I think this is same as https://github.com/smartdevicelink/sdl_core/issues/805.

shoamano83, 2018-05-01:
We didn't see the issue with 4.1 Core after https://github.com/smartdevicelink/sdl_core/issues/805 is fixed, however we started seeing the issue again when we upgraded Core to 4.3.

Possible reason is that 4.3 includes additional implementation before resumption runs, which may cause the timing issue to happen more easily. For example,
- commit https://github.com/smartdevicelink/sdl_core/pull/1106/commits/088bc6dbd4a59b4bcea423accef081cb7beae889
- commit https://github.com/smartdevicelink/sdl_core/pull/1137/commits/0a6fa22c1c4337396db122fd1ff978b38673b4ba

shoamano83, 2018-05-01:
The root cause is that `ResumeCtrlImpl::StartResumption()` / `ResumeCtrlImpl::StartResumptionOnlyHMILevel()` are triggered *after* Core notifies HMI of OnAppRegistered.

This will cause two issues:

(1) If HMI issues SDL.ActivateApp really quickly, the default HMI level may be set after SDL.ActivateApp is processed.

Since sending OnAppRegistered and processing SDL.ActivateApp run on another threads, it is possible that SDL.ActivateApp is processed before the default HMI level is set in ResumeCtrlImpl class.

(2) If HMI issues SDL.ActivateApp really quickly, the resumption timer can start after the app is activated.

The app should be removed from resumption timer during `ResumeCtrlImpl::OnAppActivated()`. However, if the app is activated quickly, this method may be called before the timer is started. This causes the HMI level to change again after the app is activated.

The fundamental fix is to trigger BC.OnAppRegistered after calling `ResumeCtrlImpl::StartResumption()` / `ResumeCtrlImpl::StartResumptionOnlyHMILevel()`.

iCollin, 2020-11-06:
Hello @shoamano83, I know it has been quite a while since you submitted this issue and PR. This issue came to light with the new resumption feature, handling errors from the HMI, [proposal 0190](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0190-resumption-data-error-handling.md). I believe this issue was actually fixed by https://github.com/smartdevicelink/sdl_core/pull/3559. If you have a chance, I would appreciate if you could confirm whether or not this issue has been fixed.

iCollin, 2021-03-05:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3559

Please feel free to reopen if you do reproduce this issue.

closed 2021-03-05 by iCollin