smartdevicelink/sdl_core #2170: [SDL 0164] Modernize Ubuntu Support

theresalech, 2018-05-03:
Proposal: [Modernize Ubuntu Support](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0164-modernize-ubuntu-support.md)

> This proposal is to add official support of the latest Ubuntu LTS Release (version 18.04 as of April 26th, 2018) to SDL Core, as well as update the minimum supported version from Ubuntu 14.04 to Ubuntu 16.04 as the former reaches the end of its support lifetime.

Review: https://github.com/smartdevicelink/sdl_evolution/issues/478

Steering Committee Decision:

> The Steering Committee agreed to accept this proposal, noting the two implementation parts as described in [jacobkeeler's comment](https://github.com/smartdevicelink/sdl_evolution/issues/478#issuecomment-384682205).

jacobkeeler, 2018-05-14:
Partially complete with the merge of #2180.

Support for Ubuntu 14.04 will be dropped in SDL Core 5.0, at that time this proposal will be complete

jacobkeeler, 2018-05-16:
Next release has been made a major release, meaning that this proposal is now fully complete

PR #2180 (2018-05-14): Modernize Ubuntu Support
Fixes #2170 (partially)

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Build and run unit tests on Ubuntu 18.04, fixing any build and test failures

### Summary
Adds support for Ubuntu 18.04

### Changelog

##### Enhancements
* Modify project to successfully build on the new Ubuntu 18.04
  * Add checks for systems that no longer support SSL3
  * Fix test failures on Ubuntu 16.04 and 18.04
  * Replace deprecated `readdir_r` with `readdir`

### Tasks Remaining:
- [x] Fix new warnings generated on Ubuntu 18.04

### CLA
- [x [... cut]
closed 2018-05-16 by jacobkeeler