smartdevicelink/sdl_core #2173: SDL shall not put remote_control appHMIType application to NONE state when user disables remote control in HMI

yang1070, 2018-05-04:
### Bug Report
SDL shall not put remote_control appHMIType application to NONE state when user disables remote control in HMI

##### Reproduction Steps
1. run sdl_core and sdl_hmi in Untuntu 16.04
2. pair an android device with the Untuntu virtual Box via Bluetooth
3. in SyncProxyTester create a new proxy via BT with isMedia=true and appHMIType has remote_control included 
4. in HMI find the app and active it
5. SPT start playing music
6. in HMI RC settings, disable remote control

##### Expected Behavior
SPT continues playing music and remains its LIMITED state

##### Observed Behavior
SPT stops playing, the app is in NONE state

##### OS & Version Information
* OS/Version: Windows10 + Ubuntu16.04
* SDL Core Version: [4.4.1]

#### This use case is very import for releasing Ford SYNC 4.0. Please prioritize the issue.

LitvinenkoIra, 2018-05-17:
@yang1070 thank you, now its clear.

LitvinenkoIra, 2018-05-21:
@yang1070 in case 

  | Current | Expected SDL core change
-- | -- | --
8 |  SDL disallow RC RPCs  GetInteriorVehicleData SetInteriorVehicleData  ButtonPress  Requests from apps | No change

what result code do we expect on the mobile side? DISALLOWED or REJECTED?

AKalinich-Luxoft, 2018-05-22:
@yang1070 Could you please clarify the following test case?

3 | HMI does not allow activate RC app | HMI show pop up “Enable remote control feature for all   mobile  apps? Please press Yes to enable remote control or No to cancel. See    Settings to disable.”       If user presses Yes, enable Remote Control.
-- | -- | --

If user press Yes, HMI should:
- enable Remote Control functionality option (in HMI settings RSDL functionality settings -> RC Functionality will be set to Allowed after that)
- send `RC.OnRemoteControlSettings(allowed = true)` notification due to option change
- activate selected app (app will be in FULL after that)

If user press No, HMI should:
- do not change Remote Control functionality option (in HMI settings RSDL functionality settings -> RC Functionality still be Disallowed)
- do not activate selected app (app still be in current HMI level)

Is this a correct HMI behaviour?

yang1070, 2018-05-23:
@AKalinich-Luxoft @askirk 
In both cases, the app will be in HMI level `FULL`. 
The difference is if user press No, all RC related RPCs are disallowed.

LitvinenkoIra, 2018-06-19:
issue will be closed after merging https://github.com/smartdevicelink/sdl_core/pull/2196

JackLivio, 2018-06-25:
@yang1070 Can you provide and documentation that shows that this is the correct behavior? If there is no documentation stating the behavior and the behavior of `putting the App into NONE when user disables remote control in HMI` has always existed on all RC releases of core, then this change will require an evolution proposal.

yang1070, 2018-06-25:
the behavior (either existing or we ask to change to)  is not documented in any previous proposals.  The asked  behavior is in the same line with other existing features, like the permission of function groups. if a user revoke the application's permission to access vehicle signals like gps, speed, etc.  The App will keep its  current HMI state and will not be put into NONE. The change is to keep app's   HMI status and not going to NONE if user disabled RC.

JackLivio, 2018-06-25:
I understand the justification for the change, but because there is no documentation for this behavior as well as the fact that RC apps have always behaved in a particular way, asking to redefine the behavior I believe will unfortunately require an evolution proposal. However we can open this up for discussion on the steering committee call for other people's opinion on the matter.

yang1070, 2018-06-25:
@JackLivio A proposal is written for this, please see PR https://github.com/smartdevicelink/sdl_evolution/pull/527

yang1070, 2018-06-26:
@JackLivio The proposal is ready for review. No matter it is as a bug fix or a new feature, there shall be a separate PR specially for this issue.  This is isolated small feature from other RC works, including rc and vehicle info plugin refactor proposal and other RC proposals that developed by Luxoft. I think the waiting for this proposal  to be reviewed and approved by SDLC shall not prevent Livio review other RC works.

jacobkeeler, 2018-08-22:
Closing as duplicate in favor of #2377

closed 2018-08-22 by jacobkeeler