smartdevicelink/sdl_core #2181: SDL Core sends HMI capabilities to Proxy before getting capabilities from HMI

t-yoshii, 2018-05-10:
### Bug Report
HMI capabilities are initially loaded from AppConfigFolder/hmi_capabilities.json file, and will be updated with [UI, TTS, VR, RC, Buttons].getCapabilities response and MixingAudioSupported response from HMI.
Current Core implementation does not wait for these responses, therefore Core can send HMI capabilities before being updated.
This behavior can cause Proxies to receive HMI capabilities which are inconsistent with those from HMI.

Currently Application Manager has "IsHMICooperating" flag. Register App Interface RPC is forced to wait this flag to become true. "IsHMICooperating" flag becomes true when HMI sends "BasicCommunication.OnReady". However, at this point, HMI capabilities are not updated yet.
I think we need an additional flag which is set after all of getCapabilities responses are received. RAI should wait for this flag instead of "IsHMICooperating".

This will ensure Core sends only updated HMI capabilities to Proxies.

##### Reproduction Steps
HMI capabilities are initialized and updated with following flow:

1. SDL Core initializes HMI capabilities from AppConfigFolder/hmi_capabilities.json.
2. SDL Core receives BasicCommunication.OnReady from HMI. SDL Core becomes RAI acceptable state.("HMI Cooperation" become OK)
3. SDL Core sends [UI, TTS, VR, RC].IsReady, Buttons.getCapabilities and MixingAudioSupport requests to HMI.
4. SDL Core receives [UI, TTS, VR, RC].IsReady response from HMI.
5. SDL Core sends [UI, TTS, VR, RC].getCapabilities to HMI if IsReady is OK.
6. SDL Core receives response for above requests and updates HMI capabilities.

If Proxy registers between 2 ~ 5, SDL Core sends HMI capabilities based on AppConfigFolder/hmi_capabilities.json.
If Proxy registers after 6, SDL Core sends updated contents of HMI capabilities based on responses from HMI.

##### Expected Behavior
HMI capabilities should be consistent

##### Observed Behavior
HMI capabilities may vary depending on the timing that RAI request is received

##### OS & Version Information
* OS/Version: all
* SDL Core Version: v4.4, v4.5
* Testing Against: Our internal devboard

##### Test Case, Sample Code, and / or Example App
PR will be created

chinlin0924, 2018-05-24:
so you connect your SDL app to SDL  core  while the SDL core and HMI are still initiating?
If you wait for SDL core and HMI are done, then connect the SDL app, does it still happen?

t-yoshii, 2018-05-25:
> so you connect your SDL app to SDL core while the SDL core and HMI are still initiating?

Yes.

> If you wait for SDL core and HMI are done, then connect the SDL app, does it still happen?

No. Issue does not occur.

E-SAITO-TMC, 2020-04-14:
@joeygrover  -san
A fix has been proposed for this bug, when will it be merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-07-24:
Hello, a similar solution to the problem was merged as part of the [persisting HMI capabilities implementation](https://github.com/smartdevicelink/sdl_core/pull/3397).

I am closing this issue as now Core does wait for the HMI to provide capabilities before replying to RAI.

closed 2020-07-24 by iCollin