smartdevicelink/sdl_core #2192: Number of threads grows when connecting/disconnecting from a video streaming app repeatedly

shoamano83, 2018-05-18:
### Bug Report

I noticed that the number of threads used by SDL Core grows if I repeat a testing cycle of connecting a video streaming app and disconnecting it. It might cause a problem on a system where maximum number of threads are tightly limited, and SDL Core runs for a long time.

##### Reproduction Steps

1. Start HMI
2. Start SDL Core on debugger:
```
$ LD_LIBRARY_PATH=. gdb ./smartDeviceLinkCore
```
3. Launch a video streaming app on the phone. Let it connect to Core through TCP transport.
4. Make the app FULL level by selecting it on HMI
5. HMI shows dialogs asking to start video and audio streamings. *Don't answer to them.*
6. The app will be automatically unregistered. (It may be registered again shortly after, but it doesn't matter.)
7. Quit the app on the phone
8. Repeat steps 3.-7. for one more time
9. Suspend SDL Core by sending a SIGTRAP from another terminal (`$ kill -TRAP pid_of_sdl_core`)
10. On the debugger terminal, run `info threads` and check the number of threads

##### Expected Behavior

There should not be duplicate thread entries in the list.

##### Observed Behavior

Duplicate thread entries are shown in the list:
- "CloseNaviAppTim" threads
- "HMI levels time" threads
- "VideoStreamSusp" threads

```
  39   Thread 0x7fffa77ee700 (LWP 12184) "HMI levels time" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  43   Thread 0x7fffa5feb700 (LWP 12193) "VideoStreamSusp" pthread_cond_wait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
  47   Thread 0x7fffa4fe9700 (LWP 12199) "HMI levels time" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  58   Thread 0x7fffa47e8700 (LWP 12228) "HMI levels time" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
  64   Thread 0x7fffa2fe5700 (LWP 12241) "VideoStreamSusp" pthread_cond_wait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
  65   Thread 0x7fffa37e6700 (LWP 12243) "CloseNaviAppTim" pthread_cond_wait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
  66   Thread 0x7fffa27e4700 (LWP 12244) "CloseNaviAppTim" pthread_cond_wait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
  69   Thread 0x7fffa1fe3700 (LWP 12247) "HMI levels time" pthread_cond_timedwait@@GLIBC_2.3.2 ()
    at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225
```

##### OS & Version Information
* OS/Version: Ubuntu 16.04 (amd64)
* SDL Core Version: commit 65c1e7eb77fa from `develop` branch
* Testing Against:
  - HMI: commit 5fc453a64b from `master` branch
  - Video streaming app: proprietary sample app running on iPhone

##### Test Case, Sample Code, and / or Example App
Debugger output (comments are added by me): [gdb_info_threads_video_streaming_app.txt](https://github.com/smartdevicelink/sdl_core/files/2015617/gdb_info_threads_video_streaming_app.txt)
Core log: [SmartDeviceLinkCore.zip](https://github.com/smartdevicelink/sdl_core/files/2015619/SmartDeviceLinkCore.zip)

shoamano83, 2018-05-18:
So far I detected two causes.

1. Type of the timer to clear used timer instances is configured to be *single-shot*, which should be *periodic*. This will be fixed with PR https://github.com/smartdevicelink/sdl_core/pull/2193. (Affected: "CloseNaviAppTim" threads)

2. `ApplicationImpl` instance is not released due to issue https://github.com/smartdevicelink/sdl_core/issues/2143. Timer threads that are tied to `ApplicationImpl` instance are therefore not released. (Affected: "HMI levels time" and "VideoStreamSusp" threads)

shoamano83, 2018-05-18:
> 2. ApplicationImpl instance is not released due to issue #2143. Timer threads that are tied to ApplicationImpl instance are therefore not released. (Affected: "HMI levels time" and "VideoStreamSusp" threads)

I think this problem introduces a random crash issue on exiting Core when `AppStopwatchImpl::WriteTime()` is triggered after `StopComponents()` is called. Since the timer thread doesn't stop, `AppStopwatchImpl::WriteTime()` may trigger AsyncRunner to run a method of PolicyHandler instance, which can be already deallocated.

E-SAITO-TMC, 2020-04-14:
@joeygrover -san
A fix(#2193) for this issue  has been proposed. Is this issue solved by # 2225? If so, when is this issue reviewed? If not, when is a fix(#2193) merged? For the next HU development of TOYOTA, it is necessary to improve the quality of SDLCore, so we believe it is necessary to modify this item.

theresalech, 2020-04-15:
Hi @E-SAITO-TMC, contents for SDL releases are determined by the Steering Committee, and the Project Maintainer works on features and bug fixes raised as high priority by Steering Committee representatives.  This particular issue has not yet been raised as high priority.  However, if Toyota would like to see this fix in the next release of SDL Core (planned for October 28, 2020), we can make note of such.  

Please confirm if this is priority for Toyota.  If so, we will include this issue in our proposed release contents, which will be presented to the Steering Committee and voted upon/finalized in June.  Thank you!

E-SAITO-TMC, 2020-04-16:
@theresalech -san Thank you for your comment. This issue is medium-priority for Toyota.

theresalech, 2020-04-20:
@E-SAITO-TMC, thank you. We have documented that this is medium priority for Toyota, and will include for Steering Committee review and vote when planning the contents for the Q4 Core release. This planning will take place in June.

iCollin, 2020-05-12:
@E-SAITO-TMC Hello, this issue has already been resolved.

The fix proposed in #2193 was already adopted in https://github.com/smartdevicelink/sdl_core/commit/bd7b7ac6f360ef5764be03c2fc9ad1c7672a0d0b. Related issue https://github.com/smartdevicelink/sdl_core/issues/2143 was also resolved by https://github.com/smartdevicelink/sdl_core/pull/2225.

To be sure, I attempted to reproduce the issue and was unable. None of the three mentioned threads are running a few seconds after the apps are disconnected.

closed 2020-05-12 by iCollin