smartdevicelink/sdl_core #2206: 'AppIconsFolderMaxSize' in smartDeviceLink.ini cannot be configured to be less than 100MB

shoamano83, 2018-05-23:
### Bug Report

The minimum value of `AppIconsFolderMaxSize` is hard-coded to `kDefaultAppIconsFolderMaxSize`, which is 100MB. OEMs cannot make this value smaller than 100MB without customizing SDL Core's code.

Reference: https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/config_profile/src/profile.cc#L1077

##### Reproduction Steps

1. Update smartDeviceLink.ini file to:
- set a dedicated folder for `AppIconsFolder`  e.g. "icons"
- set 10240 for `AppIconsFolderMaxSize`  i.e. 10KB
2. Create "icons" folder and create a dummy file of 20KB in it:
```
$ mkdir icons
$ dd if=/dev/zero of=icons/dummyfile count=20 bs=1024
```
3. Run ./start.sh
4. Launch sdl_hmi
5. Using sdl_ios' example app, connect the app to Core through TCP transport
6. Make the app in FULL level

##### Expected Behavior

The dummy file in "icons" folder should be removed as it exceeds the size specified by `AppIconFolderMaxSize`.

##### Observed Behavior

The dummy file is not removed.
```
$ ls -l icons/
合計 24
-rw-rw-r-- 1 samano samano  3906  5月 22 18:15 9999
-rw-rw-r-- 1 samano samano 20480  5月 22 18:13 dummyfile
```

Note: if I create a dummy file larger than 100MB in step 2, then the file is removed.

##### OS & Version Information
* OS/Version: Ubuntu 16.04 (amd64)
* SDL Core Version: commit 65c1e7eb77fa949cce9c07f229e2cf747f297e14 from `develop` branch
* Testing Against: sdl_hmi (commit 18460356c02e985df598b94f5a6bdbba1c451fff from `master`), example app of sdl_ios release 5.2.0 on iPhone X (iOS 11.3 beta)

##### Test Case, Sample Code, and / or Example App
Log: [SmartDeviceLinkCore.zip](https://github.com/smartdevicelink/sdl_core/files/2029225/SmartDeviceLinkCore.zip)
smartDeviceLink.ini: [smartDeviceLink_ini.zip](https://github.com/smartdevicelink/sdl_core/files/2029226/smartDeviceLink_ini.zip)

shoamano83, 2018-05-23:
Looking at the commit https://github.com/smartdevicelink/sdl_core/commit/5b58bf9b8cdd539826d209e1d6a4d233dbfa3764 I think this is an intended behavior. However, I question the reason of adding such minimum value.

- Why is such minimum value necessary?
- Why was 100MB chosen for the minimum value? (Usually an icon file is a few KB in size, so we are unlikely to hit the total size of 100MB.)
- If such minimum value is really necessary, then IMHO the document and/or the ini file should have clear description that OEMs are required to reserve 100MB or more for the icon storage.

My concern is that 100MB of storage may not be available on some embedded systems which have limited disk storage. I think OEMs should be able to choose smaller size of `AppIconsFolder` to support broader systems without modifying the code.

Note: a workaround for such systems is to configure `AppIconsAmountToRemove` to zero, so that SDL Core will not store any icon files to the folder.
https://github.com/smartdevicelink/sdl_core/blob/develop/src/components/application_manager/src/commands/mobile/set_app_icon_request.cc#L159

jacobkeeler, 2021-08-17:
Closed via #3756

PR #3756 (2021-08-17): Remove minimum value for AppIconsFolderMaxSize
Fixes #2206 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Set AppIconsFolderMaxSize to 1MB, add large file to the icons folder, then attempt to set the app icon for a connected app. Verify that the large file is deleted.

### Summary
Remove restrictions for AppIconsFolderMaxSize ini parameter

### Changelog
##### Bug Fixes
* Remove restrictions for AppIconsFolderMaxSize ini parameter

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyA [... cut]
closed 2021-08-17 by jacobkeeler