smartdevicelink/sdl_core #2215: [SDL 0158] Cloud App Transport Adapter

theresalech, 2018-05-24:
Proposal: [Cloud App Transport Adapter](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0158-cloud-app-transport-adapter.md)

> This proposal will detail a possible solution for allowing SDL-enabled cloud applications to be used in a vehicle. This proposal will cover the process of obtaining cloud application server endpoints, enabling which cloud apps should appear in the HMI app list, opening connections between Core and a cloud based app server, and how a cloud application can authenticate connected head units.

Review: smartdevicelink/sdl_evolution#493

Steering Committee Decision:
> The Steering Committee voted to accept this proposal, with the following revision: where it is mentioned in the proposal to send a status of "CONNECTING", it should say "RETRY" instead.

> The proposal .md was updated on May 24, 2018 to include the agreed upon revisions.

jordynmackool, 2019-03-13:
Accepted revisions as of 2019-03-06: https://github.com/smartdevicelink/sdl_evolution/issues/666#issuecomment-461205796

Additional revisions accepted on 2019-03-15: https://github.com/smartdevicelink/sdl_evolution/pull/689

- Make CloudAppProperties parameter `appName` non-mandatory. 
- Make CloudAppProperties parameter `appName` an array.
- Rename CloudAppProperties parameter `appName` to `nicknames`

jacobkeeler, 2019-04-17:
Closing with the merge of #2853

closed 2019-04-17 by jacobkeeler